umdlife/psdk_ros2 #18: PSDK ROS2 Wrapper Installation

mightymaxuk, 2023-10-08:
Hi, 

Thanks very much for putting the wrapper together and sharing. 

I'm trying to install ROS2 PSDK wrapper on a NVIDIA Jetson Nano, running Ubuntu 22.04. ROS2 has been installed and works. 

When installing PSDK Wrapper the build "psdk_interface" starts and finishes then when starting the "psdk_wrapper" build, after 3 mins, it hangs and doesn't change or update, this is the line it's stuck on..

**[1/2 complete] psdk_wrapper:build 23% -18.4s]**

This must be something I'm doing wrong, I've attempted the same install on Ubuntu 20.04 and got the exact same problem.

I've followed the installation instructions and made sure all the dependencies have been installed.

I would normally copy and paste the text but I cannot as the computer as crashed. 

As this has happend on both ubuntu 20.04 and 22.04, I feel this is something I'm doing wrong?

Any input would be appreciated, I've tried this 3-5 times now on each ubuntu.

Matt.

mightymaxuk, 2023-10-08:
"# Before building, check the Dependencies section and make sure you have everything installed"

where exactly are the dependencies supposed to get installed...."/psdk_ros2_ws"?

mightymaxuk, 2023-10-08:
"Use ros2 pkg list to see a list of all visible packages"

When I "ros2 pkg list" I get the long list. All dependencies listed as required for ros2 are included in the list gernerated in the terminal.

Thanks.

biancabnd, 2023-10-13:
Hi Max, thanks for your interest in the wrapper and sorry for the late reply. 
What you are reporting is a weird behavior. I don't think it has anything to do with the Ubuntu version. It could be due to the fact that you are building for an ARM  architecture on your Nvidia Jetson and not for AMD for example.  However, we have tested the wrapper on an Nvidia Jetson and we did not had this issue. We will try to see if we can reproduce this behavior. In any case you should at least have an error being reported and not simply hanging on the build process. 

> "# Before building, check the Dependencies section and make sure you have everything installed"
> 
> where exactly are the dependencies supposed to get installed...."/psdk_ros2_ws"?

No. As explained here [https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html#dependencies](https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html#dependencies), there are ROS 2 dependencies and other generic libraries that need to be installed. These are installed with apt install commands.

The only thing we sometimes need to add depending on your installation method of libopus.a (library needed by the PSDK-SDK), is a symbolic link from where you have this library installed to /usr/local/lib. This is the path that the PSDK-SDK will search for this library.

closed 2023-10-27 by vicmassy