umdlife/psdk_ros2 #21: [psdk_wrapper] DJI M30 doesn't connect

RPS98, 2023-10-26:
Hi,

Thank you very much for this work, it is very useful.

I would like to report an issue that I encountered while testing the node with the DJI M30. I have successfully tested the node with a DJI M300 RTK, but when I attempted the same test with the DJI M30, I encountered the following error:

`[ERROR] [1698328649.170769950] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 225`

I am using DJI Assistant 2 for simulation HIL and DJI E-Port for the connection of the drone.

Could you please provide guidance on how to debug this error? Any assistance would be greatly appreciated.

Thank you for your support and for this valuable work.

biancabnd, 2023-11-14:
Hi, 
You mention you managed to run the node with the DJI M300 RTK, were you using the same hardware connection via the DJI E-Port? It could be something related to how you are physically connecting the board to the drone. As this wrapper relies entirely on the DJI PSDK code, I would advice you to try to run the sample code of PSDK to ensure that you are configuring everything correctly. Once you can run the samples, you can use the same configuration to run the psdk_ros2 wrapper. You can find info about how to run the samples here: [https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/run-sample-code.html](https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/run-sample-code.html)

RPS98, 2024-01-19:
Hi,

While searching for information on the [Payload PSDK](https://developer.dji.com/doc/payload-sdk-tutorial/en/) and attempting to run the examples for different versions, I have found that although the documentation indicates that Payload SDK V3.5.0 is compatible with DJI M30, in practice, its examples fail to run, producing the same error code. Therefore, it is necessary to use a version of this wrapper compatible with Payload SDK V3.7.0, in which the PSDK examples do work. We have tried using the development branch of this version, but we have not been able to get it to compile. We hope that in the near future, the 3.7.0 version of Payload SDK or a later one can be used in this wrapper.

Thank you very much and good luck with the project

biancabnd, 2024-01-19:
Hi @RPS98, 

We have an opened PR (https://github.com/umdlife/psdk_ros2/pull/34) which migrates the wrapper to be compatible with the latest Payload-SDK version v3.8. It is still not merged into main. We need to review it but it has been tested with the M300 drone and it works. I advice you to take a look at it and you can even try this feature branch with for the M30 drone. Be aware that the configuration of the wrapper has changed slightly. Now, there are two files which need to be configured. You can find the updated documentation inside this feature branch. 

Hope it helps.

RPS98, 2024-01-19:
Hi @biancabnd,

Thank you very much, I will try it in the next few days and will update you with the results.

biancabnd, 2024-02-15:
Hi @RPS98, 
Have you managed to make the M30 work? We are using it now and everything works with just some slight configuration changes which will upload to the repo in a few weeks. Let me know if it works for you.

RPS98, 2024-02-16:
Hi @biancabnd,
The last tests I performed were with version v3.5 of the PSDK, and it didn't work. Now, with the new release 1.0 and compatibility with PSDK v3.8, I plan to test its functionality again. I hope to do it in the coming weeks. 
I will inform you of the results. 
Thank you very much for your work!

RPS98, 2024-02-19:
Hi @biancabnd,

I tried launching the psdk_ros2 again in version v1.0, and the initialization keeps failing. Could you help me with the configuration?

The current output in the terminal looks like this:

```
[INFO] [1708364080.348078918] [drone0.psdk_wrapper_node]: Creating Constructor PSDKWrapper
[INFO] [1708364080.558341396] [drone0.psdk_wrapper_node]: Configuring PSDKWrapper
[INFO] [1708364080.558369089] [drone0.psdk_wrapper_node]: Loading parameters
[INFO] [1708364080.558404085] [drone0.psdk_wrapper_node]: App name: **Hidden**
[INFO] [1708364080.558411429] [drone0.psdk_wrapper_node]: App id: **Hidden**
[INFO] [1708364080.558418080] [drone0.psdk_wrapper_node]: App key: **Hidden**
[INFO] [1708364080.558427685] [drone0.psdk_wrapper_node]: Baudrate: 921600
[INFO] [1708364080.558452184] [drone0.psdk_wrapper_node]: Using connection configuration file: /home/rafa/psdk_ros2_own_ws/install/as2_platform_dji_psdk/share/as2_platform_dji_psdk/config/link_config.json
[INFO] [1708364080.558499623] [drone0.psdk_wrapper_node]: Setting environment
[INFO] [1708364080.558530352] [drone0.psdk_wrapper_node]: Registered OSAL handler
[INFO] [1708364080.558537016] [drone0.psdk_wrapper_node]: Registered HAL handler
[INFO] [1708364080.558646976] [drone0.psdk_wrapper_node]: Loaded configuration file
[INFO] [1708364080.558655257] [drone0.psdk_wrapper_node]: Using DJI_USE_UART_USB_BULK_DEVICE
[INFO] [1708364080.558663146] [drone0.psdk_wrapper_node]: Environment has been set!
[INFO] [1708364080.558729618] [drone0.psdk_wrapper_node]: Activating PSDKWrapper
[INFO] [1708364080.558738832] [drone0.psdk_wrapper_node]: Init DJI Core...
Load configuration start.

Select link type: use_uart_and_usb_bulk_device

Config uart1 device name: /dev/ttyUSB0
Config uart2 device name: /dev/ttyACM0
Config uart2 device enable: true

Config network device name: enxf8e43b7bbc2c
Config network usb adapter vid: 0x0B95
Config network usb adapter vid: 0x1790

Config usb device vid: 0x0B95
Config usb device pid: 0x1790
Config usb bulk1 device name: /dev/usb-ffs/bulk1
Config usb bulk1 interface num: 2
Config usb bulk1 endpoint in: 0x83
Config usb bulk1 endpoint out: 0x02
Config usb bulk2 device name: /dev/usb-ffs/bulk2
Config usb bulk2 interface num: 3
Config usb bulk2 endpoint in: 0x84
Config usb bulk2 endpoint out: 0x03

Load configuration successfully.
[ERROR] [1708364087.810615822] [drone0.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 225. Retrying for 0 time. 
```

biancabnd, 2024-02-20:
Hi @RPS98, 
My colleague @stevedanomodolor can give you more details on this.

RPS98, 2024-02-22:
Hi @stevedanomodolor

I'm working on an Ubuntu 22.04 laptop, with the following connections: both 5 and 6 are connected. I have tried different link configurations, including using both bulk device and network device, but I always get error: `DJI core could not be initiated. Error code is: 225`.

Leveraging the examples test of the Payload SDK, I can successfully fly the drone using bulk configuration.

[image]

```
{
  "dji_sdk_link_config": {
    "link_available": "use_only_uart/use_uart_and_usb_bulk_device/use_uart_and_network_device",
    "link_select": "use_uart_and_network_device",
    "uart_config": {
      "uart1_device_name": "/dev/ttyUSB0",
      "uart2_device_enable": "false",
      "uart2_device_name": "/dev/ttyACM0"
    },
    "network_config": {
      "network_device_name": "enxf8e43b7bbc2c",
      "network_usb_adapter_vid": "0x0B95",
      "network_usb_adapter_pid": "0x1790"
    },
    "usb_bulk_config": {
      "usb_device_vid": "0x0B95",
      "usb_device_pid": "0x1790",
      "usb_bulk1_device_name": "/dev/usb-ffs/bulk1",
      "usb_bulk1_interface_num": "2",
      "usb_bulk1_endpoint_in": "0x83",
      "usb_bulk1_endpoint_out": "0x02",
      "usb_bulk2_device_name": "/dev/usb-ffs/bulk2",
      "usb_bulk2_interface_num": "3",
      "usb_bulk2_endpoint_in": "0x84",
      "usb_bulk2_endpoint_out": "0x03"
    }
  }
}
```

stevedanomodolor, 2024-02-23:
> Hi @stevedanomodolor
> 
> I'm working on an Ubuntu 22.04 laptop, with the following connections: both 5 and 6 are connected. I have tried different link configurations, including using both bulk device and network device, but I always get error: `DJI core could not be initiated. Error code is: 225`.
> 
> Leveraging the examples test of the Payload SDK, I can successfully fly the drone using bulk configuration.
> 
> ![Screenshot from 2024-02-22 17-31-24](https://private-user-images.githubusercontent.com/57761265/307072527-cf3d139a-9789-4a32-85c5-51936953163f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2NzYyMTcsIm5iZiI6MTcwODY3NTkxNywicGF0aCI6Ii81Nzc2MTI2NS8zMDcwNzI1MjctY2YzZDEzOWEtOTc4OS00YTMyLTg1YzUtNTE5MzY5NTMxNjNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIzVDA4MTE1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2NjdjYWE3ODEyNWVmZjBlYzJiYTFjNTJkN2E2YTEzOTU1ODMzZTUxOGYzZGFkMDA3YWM0MTQxZjNlOTk5ZmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8EbbVjXaUYSGAyRbhwTzuzUvRV1xNIbiGRhlqPvS6IY)
> 
> ```
> {
>   "dji_sdk_link_config": {
>     "link_available": "use_only_uart/use_uart_and_usb_bulk_device/use_uart_and_network_device",
>     "link_select": "use_uart_and_network_device",
>     "uart_config": {
>       "uart1_device_name": "/dev/ttyUSB0",
>       "uart2_device_enable": "false",
>       "uart2_device_name": "/dev/ttyACM0"
>     },
>     "network_config": {
>       "network_device_name": "enxf8e43b7bbc2c",
>       "network_usb_adapter_vid": "0x0B95",
>       "network_usb_adapter_pid": "0x1790"
>     },
>     "usb_bulk_config": {
>       "usb_device_vid": "0x0B95",
>       "usb_device_pid": "0x1790",
>       "usb_bulk1_device_name": "/dev/usb-ffs/bulk1",
>       "usb_bulk1_interface_num": "2",
>       "usb_bulk1_endpoint_in": "0x83",
>       "usb_bulk1_endpoint_out": "0x02",
>       "usb_bulk2_device_name": "/dev/usb-ffs/bulk2",
>       "usb_bulk2_interface_num": "3",
>       "usb_bulk2_endpoint_in": "0x84",
>       "usb_bulk2_endpoint_out": "0x03"
>     }
>   }
> }
> ```

@RPS98 Can you send the configuration you used in the sample test that worked for you?, 

The json you send should work for a the network configuration. One way to ensure you have connection right is to make sure that  this network interface `enxf8e43b7bbc2c` exists on your machine and also that the  when you unplug and plug the the eport device, that the /dev/ttyUSB0 shows. If that is not the case, it is a matter of connection.

bitcurious, 2024-04-04:
> Hi @biancabnd,
>
> I tried launching the psdk_ros2 again in version v1.0, and the initialization keeps failing. Could you help me with the configuration?
>
> The current output in the terminal looks like [...]

Hi @
I have working on M300 and faced same issue. But when I power-cycle the drone ones and again run the code it get initiatedand I was getting the data.
My configuration was as follows 
{
    "dji_sdk_link_config": {
        "link_available": "use_only_uart/use_uart_and_usb_bulk_device/use_uart_and_network_device",
        "link_select": "use_only_uart",
        "uart_config": {
            "uart1_device_name": "/dev/ttyUSB1",
            "uart2_device_enable": "false",
            "uart2_device_name": "/dev/dji_advanced_sensing"
        },
        "network_config": {
            "network_device_name": "enxf8e43b7bbc2c",
            "network_usb_adapter_vid": "0x0B95",
            "network_usb_adapter_pid": "0x1790"
        },
        "usb_bulk_config": {
            "usb_device_vid": "0x0B95",
            "usb_device_pid": "0x1790",
            "usb_bulk1_device_name": "/dev/usb-ffs/bulk1",
            "usb_bulk1_interface_num": "2",
            "usb_bulk1_endpoint_in": "0x83",
            "usb_bulk1_endpoint_out": "0x02",
            "usb_bulk2_device_name": "/dev/usb-ffs/bulk2",
            "usb_bulk2_interface_num": "3",
            "usb_bulk2_endpoint_in": "0x84",
            "usb_bulk2_endpoint_out": "0x03"
        }
    }
}

RPS98, 2024-04-11:
Hi @bitcurious,

Thank you for sharing your experience. With that configuration, were you able to get live image view?

bitcurious, 2024-04-12:
Hi @RPS98 I haven't tried the camera and liveview module yet but telemetry and flight control module is working fine with it. I will try it in next week and will update you here.

Also please let me know if we need to change anything to initialize camera and liveview module like DO we need to add the payload information (camera module name) anywhere in configuration?

stevedanomodolor, 2024-04-12:
@RPS98 @bitcurious, We were able to connect both the M30 and M350 with camera streaming. The main problem most people encounter is mainly with the bulk/network configuration which is responsible for the camera streaming. The serial communication does not require any special configuration if you have the FTDI and follow the tutorial as stated in the tutorial from DJI https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/jetson-nano.html. 

If i may ask what computer are you using? @RPS98 ?

bitcurious, 2024-04-12:
@stevedanomodolor What is FTDI ?

stevedanomodolor, 2024-04-12:
> @stevedanomodolor What is FTDI ?

It is is a type of USB-to-serial interface cable that allows to convert usb to uart signal, it is the USB to TTL Serial Module	in the image below. Take into consideration that this is if you use the eport development kit from dji. 
[image]

RPS98, 2024-04-12:
> @RPS98 @bitcurious, We were able to connect both the M30 and M350 with camera streaming. The main problem most people encounter is mainly with the bulk/network configuration which is responsible for [...]

I used the connection setup from the tutorial you sent, with the Payload and TTL connections.

- The serial connection (_use_only_uart_) works on all three models, M300, M30, and M350, allowing me to fly without camera streaming. 
- The Bulk connection (_use_uart_and_usb_bulk_device_) works with the M300, enabling camera streaming. However, it doesn't work with the M30 and M350. 
- The network connection (_se_uart_and_network_device_) doesn't work with any of the three.

I understand the issue is with configuration, not hardware, when using network connection. I've been conducting tests with my laptop (Ubuntu 22.04), successfully connecting with the M300 and detecting the network on the M350, but unable to initialize Camera Manager. 

I plan to continue these tests with an Nvidia Jetson once it's available.

bitcurious, 2024-04-12:
@stevedanomodolor Okay got it. Yes I'm using E-port development module and USB to TTL serial module as well. 

So I have one question which link configuration from following one should use to get camera data as well if they are using E-port Development board and TTL module?
1. use_only_uart 
2. use_uart_and_usb_bulk_device
3. use_uart_and_network_device"

stevedanomodolor, 2024-04-12:
> @stevedanomodolor Okay got it. Yes I'm using E-port development module and USB to TTL serial module as well.
>
> So I have one question which link configuration from following one should use to get [...]

Depends on which configuration you decide to do. In the bulk/network configuration of the nvidia for example, you have the options to choose, enable_rndis of enable_bulk, with the rndis configuration for examples, you would use the network configuration with the correct ntwork interface based on your computer.

RPS98, 2024-04-17:
Hi @bitcurious 

> @stevedanomodolor @RPS98 Currently, I am using the DJI Matrice 300 (M300) drone with the Onboard SDK (OSDK) extension module and the H20T camera. However, I am facing issues with initializing the camera module when setting it to true, which prevents the wrapper from running successfully.
> 
> Additionally, I have configured the system to use 'use_uart_and_usb_bulk_device'.

I'm not sure if the Onboard SDK (OSDK) extension module is compatible with the PSDK. I am using the [DJI E-Port Development Kit](https://store.dji.com/product/dji-e-port-development-kit?vid=141171&set_region=US&from=site-nav).

> 
> Furthermore, when I ran the command `lsusb`, I obtained the following output: `Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC`.

The UART is used by the PSDK for basic flight functions. To obtain images, the bulk or network connection is necessary. So, you should have two devices connected. The OSDK extension module has the FTDI and USB-A connection. In my case, when running `lsusb`:

```
Bus 003 Device 008: ID 2ca3:001f DJI Technology Co., Ltd. e1e
Bus 003 Device 009: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
```

I hope it helps you

stevedanomodolor, 2024-04-17:
> @stevedanomodolor @RPS98 Currently, I am using the DJI Matrice 300 (M300) drone with the Onboard SDK (OSDK) extension module and the H20T camera. However, I am facing issues with initializing the camera module when setting it to true, which prevents the wrapper from running successfully.
> 
> Additionally, I have configured the system to use 'use_uart_and_usb_bulk_device'.
> 
> In regards to additional information, when I ran the command `ps -aux | grep startup_bulk`, I received the following output: `root 401 0.0 0.0 9236 748 pts/6 S+ 01:53 0:00 grep --color=auto startup_bulk`.
> 
> Furthermore, when I ran the command `lsusb`, I obtained the following output: `Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC`.
> 
> Based on the provided VID and PID, the VID is `0403` and the PID is `6001` is used in usb_bulk_config.
> 
> **Please advise on the required configuration to successfully obtain the camera stream.**
> 
> ![Screenshot from 2024-04-17 14-24-08](https://private-user-images.githubusercontent.com/78998290/323154180-2ae6a9af-3ec9-41b9-935e-fc9bbe9f781d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNDc5NDEsIm5iZiI6MTcxMzM0NzY0MSwicGF0aCI6Ii83ODk5ODI5MC8zMjMxNTQxODAtMmFlNmE5YWYtM2VjOS00MWI5LTkzNWUtZmM5YmJlOWY3ODFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDA5NTQwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmZTI1MmQ4NmJkNzNjYzQyZjlhY2Y5YzE3NmY0YWU0OTNiNDg2MTJkZmM4NDgxZTAyYTdmMDZlOWMzMTZlZDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zLVZaOUpXGGPkn6lG-VoyVEulCDgFy5LvVFC5N6GdVI)

With the **m300**, using the osdk module expansion module and with the appropriate connection, the psdk_ros2 library should work. Can you share a schematic of your hardware connection from the drone to the osdk module and to the computer?

bitcurious, 2024-04-18:
@stevedanomodolor sure 
here is my connection with M300 using OSDK expansion module.

[image]

[image]

[image]

bitcurious, 2024-04-18:
@stevedanomodolor also I have connected the **OSDK USB2.0 Interface** and found some following observation.

I don't think the 3-pin connector (Uart) and the USB connection can be active at the same time

No 3-pin connection(Uart), USB connected to Jetson USB-C on the power input USB-C:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2ca3:001f DJI e1e
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
USB Cable with FTDI interface AND the USB connection:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2ca3:001f DJI e1e
Bus 001 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
But as soon as I plug the other end of the FTDI cable (which is the 3-pin) into the OSDK:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
It looks like the UART supersedes the USB/DJI e1e connection

Please let me know how both 3-pin connector (Uart) and the USB connection can be active at the same time

bitcurious, 2024-04-19:
@stevedanomodolor @RPS98 @biancabnd 
After some trial and error, I discovered that the issue was with the USB port. After some adjustments, I was able to resolve the problem. Upon running `lsusb`, I obtained the following output:
```
- Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
- Bus 001 Device 005: ID 2ca3:001f DJI e1e
```
With this configuration, the camera module initialized successfully, and I started receiving data on the camera stream topic.

biancabnd, 2024-04-19:
Hi @bitcurious, 
I'm glad that you managed to make it work. We will try to improve the documentation so that it gives some more details about the actual hardware connection, although this is always custom to each end user. But thanks for your feedback.

bitcurious, 2024-04-23:
@RPS98 Since you are getting camera stream with M300 and not with M30 and you are using E-port for both right?
can you check the status of **USB ID Switch** it should be on Host mode for M30 drone and for M300 it should on device mode as DJI mention in their documentation you can see the sequence no 10. 
[image]

biancabnd, 2024-04-24:
@RPS98 , @bitcurious have you managed to correctly initialize all modules and obtain the main camera streaming on the DJI M30 drone? If yes, could you tell me which dev kit have you used for the connection (E-Port dev kit/ SDK round ribbon cable)?

bitcurious, 2024-04-24:
@biancabnd I haven't had the opportunity to try the psdk_wrapper on M30 yet. At the moment, I only have access to the M300. probably I will get the access to the M30 in next month.

biancabnd, 2024-04-24:
Thank you for the feedback. We are trying to improve the documentation and give more details on how the HW configuration needs to be done.

AndreaSanchini, 2025-05-29:
Hi @javilinos, thank you for your work. 
I'm currently working with a Jetson Orin Nano and Djji M350. I followed https://aerostack2.github.io/_03_aerial_platforms/_dji_psdk/index.html and I was able to configure the usb bulk. However when trying to run the wrapper only `use_uart_and_network_device` with the `l4tbr0` seems to work, while when using `use_uart_and_usb_bulk_device` camera manager init fails. Did you manage to start the wrapper using both options?