umdlife/psdk_ros2 #22: psd_wrapper package hangs jetson on build

abeldixon, 2023-11-08:
OS: Ubuntu 18.04 with ROS 2 Humble docker container

I have installed the required dependencies within the container, but during the build process the psdk_wrapped freezes the NVIDIA Jetson.

What might be the potential issues causing this ?

What are the system requirement for running this wrapper ? Was this wrapper also tested in docker container ?
Also It seems there isn't an official Ubuntu 22.04 version available for NVIDIA Jetson, so how the setup has to be done ?

vicmassy, 2023-11-13:
Hi @abeldixon, to answer few of your questions. 
The package has been tested over amd64 and arm64 architectures. We usually deploy using it with docker indeed. The base image though it has to be Ubuntu22 or simply `ros:humble` 
For the Jetson, so far the latest is Ubuntu20.04 indeed, but you can run a docker with Ubuntu22 without any issues. Moreover. soon (December) they will release the new Jetpack with Ubuntu 22.04. 

So what I would do is try to build the package with the Docker Image `ros:humble-ros-base` from https://hub.docker.com/_/ros/
It should work out of the box. Make sure to follow the installation instructions, in here https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html

Hope it helps.

abeldixon, 2023-11-14:
Thank you for the response @vicmassy, so we will have to do cross compilation from amd64 to arm64 to develop locally within our system right ? Because building the ros package within the docker is causing jetson to run out of memory, is there any walkthrough tutorials for the doing the cross builds ?

vicmassy, 2023-11-14:
If you want you can use the already pre-built binaries so you don't need to compile the wrapper. You can find them here: https://github.com/umdlife/psdk_ros2/releases Take the correct architecture and to install it use `sudo dpkg -i NAME.deb`