umdlife/psdk_ros2 #41: Step by step initialization

santiago-tapia, 2024-01-18:
## Feature request

#### Feature description
The initialization at psdk_wrapper.cpp line #97 just fails on the failure on any of the module initializations, it could be better to init the modules one by one and add a particular message to report where to look for the failure. Actually it would nice to launch the wrapper even if one of module fails to init.

#### Implementation considerations
The implementation is trivial, just calling to each of the initialization functions in a separate "if". But, could it be possible to continue the initialization if an optional module fails?

#### Are you willing to submit a pull request to implement this change?
Well, it depends on the answer to the previous question, my opinion is that the initialization should continue on the failure of any optional module, but I am not sure.

biancabnd, 2024-01-18:
Hi @santiago-tapia,
Thank you for your comment and for using the wrapper. From what I understand, you are mentioning here two issues:

1) Adding more granularity on the initialization process of the modules and knowing which one is failing. I think this is partly present in the current code as each module initialization method has an error message inside. It is however true, that once one module fails, the entire process is stopped. 

2) Define two categories of modules, one mandatory and one optional and continue the wrapper initialization if all mandatory modules are up and running. This is indeed a nice feature to have as for some users, some modules might not be needed. We might need to check though that there are no dependencies among modules in order to correctly implement this. Nonetheless, the modules we are currently running are pretty basic in my opinion. Can you give me an example of which one would you consider as optional, just to better understand your motivation on this?

santiago-tapia, 2024-01-18:
Hi @biancabnd ,

1. Great, sorry, I have overlooked it. That's OK then.
2. We propose this point just like an interesting general feature. But, actually, we thought about it because we were having problems with the camera init method, _init_camera_manager_, it was failing while we were testing some other things about flight control. So we had to comment it out to continue. Since the camera was optional in our circunstances we thought it was an interesting feature to be able to run the wrapper without it (and without to comment out any line).

RPS98, 2024-01-23:
Hi @biancabnd,

Another improvement should be to store the return error code of `DjiCameraManager_Init` and use it in the ROS error. Applying this to all modules would be very useful for debugging errors.

https://github.com/umdlife/psdk_ros2/blob/0fdac3f6d74e4f8c3a7c4730e89de689a3d152d6/psdk_wrapper/src/modules/camera.cpp#L27-L31

biancabnd, 2024-01-25:
@RPS98 That sounds good to me! I will try to add these features together with the migration to psdk v3.8

biancabnd, 2024-02-15:
Hi @santiago-tapia, 
The latest version of the `psdk_ros2` wrapper (1.0.0) contains the feature of setting modules as non-mandatory. Please check the updated documentation or directly the `psdk_params.yml` file. I hope this is useful for your particular situation. I will close this issue now.

closed 2024-02-15 by biancabnd