umdlife/psdk_ros2 #42: Camera Manager Fails Initialization

RPS98, 2024-01-22:
**Tell us about your environment**

* **Ubuntu version:** 22.04
* **Docker Version:**
* **ROS/ROS2 version:** Humble
* **Commit hash:** [Main Release v0.0.2](c9629d50020c2b4e63e81084d129263d406b87aa)
* **Aircraft:** DJI M300 RTK
* **Aircraft devices:** None
* **Firmware:** v09.00.0503
* **Connection:** E-port Development Kit

#### Steps to reproduce issue
Follows [Getting Started tutorial](https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html)

#### Expected behavior
Launch node

#### Actual behavior
Fails initialization

#### Additional information
Screen output:
```
rafa@rafa:~/psdk_ros2_ws$ ros2 launch psdk_wrapper wrapper.launch.py
[INFO] [launch]: All log files can be found below /home/rafa/.ros/log/2024-01-22-12-19-53-759617-rafa-69121
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [psdk_wrapper_node-1]: process started with pid [69132]
[psdk_wrapper_node-1] [INFO] [1705922393.886475966] [wrapper.psdk_wrapper_node]: Creating Constructor PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1705922394.106358892] [wrapper.psdk_wrapper_node]: Configuring PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1705922394.106389113] [wrapper.psdk_wrapper_node]: Loading parameters
[psdk_wrapper_node-1] [INFO] [1705922394.106430706] [wrapper.psdk_wrapper_node]: App name: **Hide**
[psdk_wrapper_node-1] [INFO] [1705922394.106435790] [wrapper.psdk_wrapper_node]: App id: **Hide**
[psdk_wrapper_node-1] [INFO] [1705922394.106440940] [wrapper.psdk_wrapper_node]: App key: **Hide**
[psdk_wrapper_node-1] [INFO] [1705922394.106448067] [wrapper.psdk_wrapper_node]: Baudrate: 921600
[psdk_wrapper_node-1] [INFO] [1705922394.106452728] [wrapper.psdk_wrapper_node]: Hardware connection: DJI_USE_UART_AND_USB_BULK_DEVICE
[psdk_wrapper_node-1] [INFO] [1705922394.106460606] [wrapper.psdk_wrapper_node]: Uart dev 1: /dev/ttyUSB0
[psdk_wrapper_node-1] [INFO] [1705922394.106465970] [wrapper.psdk_wrapper_node]: Uart dev 2: /dev/ttyACM0
[psdk_wrapper_node-1] [INFO] [1705922394.106494175] [wrapper.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1705922394.106522775] [wrapper.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1705922394.106528430] [wrapper.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] [INFO] [1705922394.106531375] [wrapper.psdk_wrapper_node]: Using DJI_USE_UART_AND_USB_BULK_DEVICE
[psdk_wrapper_node-1] [INFO] [1705922394.106537109] [wrapper.psdk_wrapper_node]: Environment has been set!
[psdk_wrapper_node-1] [INFO] [1705922394.106540918] [wrapper.psdk_wrapper_node]: Initializing ROS publishers
[psdk_wrapper_node-1] [INFO] [1705922394.119040199] [wrapper.psdk_wrapper_node]: Creating subscribers
[psdk_wrapper_node-1] [INFO] [1705922394.121143990] [wrapper.psdk_wrapper_node]: Creating services
[psdk_wrapper_node-1] [INFO] [1705922394.135900234] [wrapper.psdk_wrapper_node]: Activating PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1705922394.135918203] [wrapper.psdk_wrapper_node]: Init DJI Core...
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyACM0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyACM0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyACM0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyACM0': Operation not permitted
[psdk_wrapper_node-1] [INFO] [1705922399.571133205] [wrapper.psdk_wrapper_node]: Activating ROS elements
[psdk_wrapper_node-1] [INFO] [1705922399.571222133] [wrapper.psdk_wrapper_node]: Initiating telemetry...
[psdk_wrapper_node-1] [INFO] [1705922399.571230034] [wrapper.psdk_wrapper_node]: Initiating flight control module...
[psdk_wrapper_node-1] [INFO] [1705922399.580274733] [wrapper.psdk_wrapper_node]: Initiating camera manager...
[psdk_wrapper_node-1] [ERROR] [1705922399.609467516] [wrapper.psdk_wrapper_node]: Could not initialize camera manager.
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/wrapper/psdk_wrapper_node'
```

Connection:
[image]

biancabnd, 2024-01-23:
Hi @RPS98, 

Thi is not a bug. In this case the error is clearly mentioned in the logs you are receiving. The process from where you are running the psdk ros2 wrapper needs to have enough privileges to manage the devices (in this case the /dev/ttyUSB0) through which you communicate with the drone. So either run it as root, or manage the permissions of the user through which you are running the psdk ros2 wrapper.

RPS98, 2024-01-23:
Hi @biancabnd,

I don't think this is the problem, since the system has write permissions on those devices, but these messages are produced because the node does not check permissions before attempting to overwrite them. We shouldn't launch nodes with root permissions; that's why they appear. The code continues to run despite these failed attempts. 

By commenting out the Camera Manager initialization line, the code works correctly, and I can fly the drone. Therefore, I'm not sure if the issue is related to permissions, which shouldn't be the case since the system has write permissions and PSDK examples work correctly, or if the issue is related to something else.

What is your testing setup?

RPS98, 2024-01-25:
Hi @biancabnd,

I am encountering error code 227 when calling `DjiCameraManager_Init`. I believe I can convert this error code into a string using `dji_error.h`, but currently, I am unsure how to do so. Could you assist me with this?

Thank you very much for your help.

biancabnd, 2024-01-30:
Hi @RPS98, 
This error corresponds to DJI_ERROR_SYSTEM_MODULE_CODE_INVALID_PARAMETER. 
I assume you do have a camera attached to the gimbal, right? Just for reference, we have tested this wrapper with both H20 and H20T payloads.

RPS98, 2024-02-02:
Hi @biancabnd,

Yes, it has mounted the camera H20T. Is there any way to set up the node without any camera device mounted?

Also, I managed to launch the node correctly with the DJI M300 RTK, but the DJI M350 RTK shows the error "Camera manager fails with error code 227". I'm using the same hardware with both drones and the latest version of the `feat/upgrade-to-psdk-v3.8` branch. I also tried the `main` branch, and the same error persists.

Output:
```
[psdk_wrapper_node-1] [INFO] [1706897708.615641641] [drone0.psdk_wrapper_node]: Creating Constructor PSDKWrapper                                                                                             [5/12]
[psdk_wrapper_node-1] [INFO] [1706897708.834981203] [drone0.psdk_wrapper_node]: Configuring PSDKWrapper                                                                                                            
[psdk_wrapper_node-1] [INFO] [1706897708.835014037] [drone0.psdk_wrapper_node]: Loading parameters                                                                                                                 
[psdk_wrapper_node-1] [INFO] [1706897708.835070513] [drone0.psdk_wrapper_node]: App name: **Hidde**                                                                                                                     
[psdk_wrapper_node-1] [INFO] [1706897708.835076266] [drone0.psdk_wrapper_node]: App id: **Hidde**                                                                                                                     
[psdk_wrapper_node-1] [INFO] [1706897708.835093202] [drone0.psdk_wrapper_node]: App key: **Hidde**                                                                                           
[psdk_wrapper_node-1] [INFO] [1706897708.835101058] [drone0.psdk_wrapper_node]: Baudrate: 921600
[psdk_wrapper_node-1] [INFO] [1706897708.835106607] [drone0.psdk_wrapper_node]: Using connection configuration file: /home/rafa/as2_projects/project_dji_psdk/real_config/link_config.json
[psdk_wrapper_node-1] [INFO] [1706897708.835141446] [drone0.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1706897708.835161892] [drone0.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1706897708.835167900] [drone0.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] [INFO] [1706897708.835268068] [drone0.psdk_wrapper_node]: Loaded configuration file 
[psdk_wrapper_node-1] [INFO] [1706897708.835273891] [drone0.psdk_wrapper_node]: Using DJI_USE_UART_USB_BULK_DEVICE
[psdk_wrapper_node-1] [INFO] [1706897708.835280174] [drone0.psdk_wrapper_node]: Environment has been set! 
[psdk_wrapper_node-1] [INFO] [1706897708.835286309] [drone0.psdk_wrapper_node]: Init DJI Core...
[psdk_wrapper_node-1] Load configuration start.                                                          
[psdk_wrapper_node-1]                                                                                    
[psdk_wrapper_node-1] Select link type: use_uart_and_usb_bulk_device                        
[psdk_wrapper_node-1]                                                                                    
[psdk_wrapper_node-1] Config uart1 device name: /dev/ttyUSB0                                
[psdk_wrapper_node-1] Config uart2 device name: /dev/ttyACM0                                
[psdk_wrapper_node-1] Config uart2 device enable: true                                                                                                                                                             
[psdk_wrapper_node-1]                                                                                                                                                                                              
[psdk_wrapper_node-1] Config network device name: enxf8e43b7bbc2c                                                                                                                                                  
[psdk_wrapper_node-1] Config network usb adapter vid: 0x0B95                                                                                                                                                       
[psdk_wrapper_node-1] Config network usb adapter vid: 0x1790                                                                                                                                                       
[psdk_wrapper_node-1]                                                                                                                                                                                              
[psdk_wrapper_node-1] Config usb device vid: 0x0B95                                                                                                                                                                
[psdk_wrapper_node-1] Config usb device pid: 0x1790                                                                                                                                                                
[psdk_wrapper_node-1] Config usb bulk1 device name: /dev/usb-ffs/bulk1                                                                                                                                             
[psdk_wrapper_node-1] Config usb bulk1 interface num: 2                                                                                                                                                            
[... cut ...]
[psdk_wrapper_node-1] sh: 1: cannot create /proc/sys/net/core/rmem_max: Permission denied
[psdk_wrapper_node-1] [INFO] [1706897718.854116883] [drone0.psdk_wrapper_node]: Initiating telemetry...
[psdk_wrapper_node-1] [INFO] [1706897718.854167703] [drone0.psdk_wrapper_node]: Initiating flight control module...
[psdk_wrapper_node-1] [INFO] [1706897718.918126021] [drone0.psdk_wrapper_node]: Initiating camera manager...
[psdk_wrapper_node-1] [ERROR] [1706897718.949912798] [drone0.psdk_wrapper_node]: Could not initialize camera manager. Error code: 227
```

biancabnd, 2024-02-08:
Hi @RPS98,
I will add a new parameter to define optional/mandatory modules so that you can skip the camera initialization. However, it is indeed an issue that the camera cannot be initialized with the M350. We will try to use an M350 soon and we'll debug the issue, because at this moment I cannot really tell what is happening. There might be some differences in the DJI PSDK libs when using this type of drone.

RPS98, 2024-02-19:
Hi @biancabnd,
I've tried release v1.0 and it still has the same initialization issue with the camera module and liveview. However, the rest of the systems work perfectly, and the drone flies without issues. Thank you for those configuration parameters.
The error that appears is the following:

```
Load configuration successfully.
sh: 1: cannot create /proc/sys/net/core/rmem_default: Permission denied
sh: 1: cannot create /proc/sys/net/core/rmem_max: Permission denied
[INFO] [1708341654.438697944] [drone0.psdk_wrapper_node]: Initiating telemetry...
[INFO] [1708341654.438743389] [drone0.psdk_wrapper_node]: Initiating flight control module...
[INFO] [1708341654.477004680] [drone0.psdk_wrapper_node]: Initiating camera manager...
[ERROR] [1708341654.516396839] [drone0.psdk_wrapper_node]: Could not initialize camera manager. Error code: 227
[INFO] [1708341654.516411011] [drone0.psdk_wrapper_node]: Initiating gimbal manager...
[INFO] [1708341654.516419325] [drone0.psdk_wrapper_node]: Initiating liveview module...
[ERROR] [1708341654.516436702] [drone0.psdk_wrapper_node]: Could not initialize liveview module. Error code: 236
[INFO] [1708341654.516462663] [drone0.psdk_wrapper_node]: Initializing ROS publishers
[INFO] [1708341654.534749963] [drone0.psdk_wrapper_node]: Creating subscribers
[INFO] [1708341654.537748243] [drone0.psdk_wrapper_node]: Creating services
[INFO] [1708341654.558631366] [drone0.psdk_wrapper_node]: Activating ROS elements
[INFO] [1708341654.558668298] [drone0.psdk_wrapper_node]: Publishing static transforms
[INFO] [1708341655.415654299] [drone0.psdk_wrapper_node]: Setting local altitude reference to: 100.099998
[INFO] [1708341663.341308869] [drone0.psdk_wrapper_node]: Setting up camera streaming for payload index 1 and camera source 0. Output decoded: 0
[INFO] [1708341663.341341726] [drone0.psdk_wrapper_node]: Starting streaming...
[ERROR] [1708341663.341354562] [drone0.psdk_wrapper_node]: Failed to set-up the decoder
```

biancabnd, 2024-02-19:
Hi @RPS98, 
As specified in the dji_error.h in the DjiErrorCode enum, the 227 error code is: DJI_ERROR_SYSTEM_MODULE_CODE_INVALID_PARAMETER 
Unfortunately we cannot reproduce it yet as we don't have this HW available. Once, we do I can let you know our findings. Meanwhile, you can check DJI forums to see if there is anything reported on the matter.

Janphr, 2024-03-22:
Hey,
I'm also trying to get video streaming running on a M350. I get successful response from service call for fpv, but nothing is being published. Main camera doesn't seem to work at all.
```
[INFO] [1711100106.377158769] [dji.psdk_wrapper_node]: Initiating telemetry...
[INFO] [1711100106.377248937] [dji.psdk_wrapper_node]: Initiating flight control module...
[INFO] [1711100106.412146891] [dji.psdk_wrapper_node]: Initiating camera manager...
[INFO] [1711100108.031476724] [dji.psdk_wrapper_node]: Checking connected payloads...
[ERROR] [1711100110.035057348] [dji.psdk_wrapper_node]: Get mounted position 1 camera's type failed, error code: 225
[INFO] [1711100110.035152293] [dji.psdk_wrapper_node]: Initiating gimbal manager...
[INFO] [1711100110.035171646] [dji.psdk_wrapper_node]: Initiating liveview module...
[INFO] [1711100111.052503740] [dji.psdk_wrapper_node]: Initializing ROS publishers
[INFO] [1711100111.088805564] [dji.psdk_wrapper_node]: Creating subscribers
[INFO] [1711100111.093093458] [dji.psdk_wrapper_node]: Creating services
[INFO] [1711100111.163672196] [dji.psdk_wrapper_node]: Activating ROS elements
[INFO] [1711100111.163752326] [dji.psdk_wrapper_node]: Publishing static transforms
[INFO] [1711100111.163774919] [dji.psdk_wrapper_node]: Initiating HMS...
[INFO] [1711100112.355622510] [dji.psdk_wrapper_node]: Setting local altitude reference to: 100.099998
[INFO] [1711100158.805246922] [dji.psdk_wrapper_node]: Setting up camera streaming for payload index 7 and camera source 0. Output decoded: -1632502768
[INFO] [1711100158.805310125] [dji.psdk_wrapper_node]: Starting streaming...
[INFO] [1711100160.144057382] [dji.psdk_wrapper_node]: Successfully started the camera streaming.
[ERROR] [1711100167.601669616] [dji.psdk_wrapper_node]: Settinh mounted position 7 camera's work mode as record-video mode failed, error code :224
[INFO] [1711100190.103740996] [dji.psdk_wrapper_node]: Setting up camera streaming for payload index 7 and camera source 0. Output decoded: -1632502768
[INFO] [1711100190.103796626] [dji.psdk_wrapper_node]: Stopping camera streaming...
[INFO] [1711100190.618970968] [dji.psdk_wrapper_node]: Successfully stopped camera streaming.
[INFO] [1711100202.806811296] [dji.psdk_wrapper_node]: Setting up camera streaming for payload index 1 and camera source 0. Output decoded: -1632502768
[INFO] [1711100202.806874629] [dji.psdk_wrapper_node]: Starting streaming...
[ERROR] [1711100203.318875350] [dji.psdk_wrapper_node]: Failed to start camera streaming, error code: 256.
[INFO] [1711100257.769678734] [dji.psdk_wrapper_node]: Setting up camera streaming for payload index 7 and camera source 0. Output decoded: -1632502768
[INFO] [1711100257.769732585] [dji.psdk_wrapper_node]: Starting streaming...
[INFO] [1711100258.315794467] [dji.psdk_wrapper_node]: Successfully started the camera streaming.
```
I have no payload camera connected. Just wanted to try the front camera. Also, are you planning to add support for the stereo cameras?

biancabnd, 2024-03-26:
Hi @Janphr, 
For now, we are only retrieving the streaming from the main cameras (attached to the gimbal) and the FPV camera. The perception module which provides the access to the stereo cameras is not integrated at this moment in the wrapper. To be honest, this is not our main priority at this moment so it's unclear when we will support this. If you need it urgentlly, please feel free to open a new PR with the integration of this module. (dji_perception as referred to in the DJI Payload-SDK)

biancabnd, 2024-04-24:
@RPS98 have you managed to initialize the camera manager module with your set-up (DJI M350, E-Port Dev Kit)?

RPS98, 2024-04-24:
Hi @biancabnd, 
I haven't managed to get images with the DJI M350 yet. Although I haven't had time to continue investigating either. 
If you or someone else has managed to launch everything correctly on the DJI M350 and get images, documenting the hardware connection and software configuration would be very helpful, and the issue could be closed

biancabnd, 2024-04-24:
Thank you for the feedback. Yes, we have managed to retrieve the streaming with the M350. We are trying to improve the documentation and give more details, as much as we can, on how the HW configuration needs to be done. However, it is very much dependent on each user's set-up, board type so it cannot be easily generalized.

bonchan, 2024-06-17:
> e DJI M300 RTK, but the DJI M350 RT

hi there. how did you manage to run the node correctly?

javilinos, 2025-05-22:
Hi @biancabnd and @everyone. We have managed to put the psdk ros2 wrapper's sensing and control to work in the M300, M350 and M30T in both AGX Orin and NX Orin, also with and without H20T. You can find instructions about out software and hardware configuration in https://aerostack2.github.io/_03_aerial_platforms/_dji_psdk/index.html. If you have any doubts, please mention myself or @RPS98, we are still working on this documentation and we will be pleased to answer all of your doubts.

biancabnd, 2025-05-22:
Hi @javilinos! That is great news! Thank you for continuing developing and improving the wrapper. As an update, I am no longer affiliated with the company mantaining this open sourced repository, thus I cannot actively maintain it.