umdlife/psdk_ros2 #43: DJI PSDK v3.8 Fails Initialization

RPS98, 2024-01-22:
**Tell us about your environment**

* **Ubuntu version:** 22.04
* **Docker Version:**
* **ROS/ROS2 version:** Humble
* **Commit hash:** [feat/upgrade-to-psdk-v3.8](72cac881cc38d8086b4e5aac80f5f0bba5a2c7ee)
* **Aircraft:** DJI M300 RTK
* **Aircraft devices:** None
* **Firmware:** v09.00.0503
* **Connection:** E-port Development Kit

#### Steps to reproduce issue
Follows [Getting Started tutorial](https://github.com/umdlife/psdk_ros2/blob/feat/upgrade-to-psdk-v3.8/docs/documentation/GettingStarted.md)

#### Expected behavior
Launch node

#### Actual behavior
Fails initialization

#### Additional information
Screen output:
```
rafa@rafa:~/psdk_ros2_3_8_ws$ ros2 launch psdk_wrapper wrapper.launch.py
[INFO] [launch]: All log files can be found below /home/rafa/.ros/log/2024-01-22-12-11-00-846514-rafa-56701
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [psdk_wrapper_node-1]: process started with pid [56712]
[psdk_wrapper_node-1] [INFO] [1705921860.969902844] [wrapper.psdk_wrapper_node]: Creating Constructor PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1705921861.188747889] [wrapper.psdk_wrapper_node]: Configuring PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1705921861.188777887] [wrapper.psdk_wrapper_node]: Loading parameters
[psdk_wrapper_node-1] [INFO] [1705921861.188809456] [wrapper.psdk_wrapper_node]: App name: **Hide**
[psdk_wrapper_node-1] [INFO] [1705921861.188814244] [wrapper.psdk_wrapper_node]: App id: **Hide**
[psdk_wrapper_node-1] [INFO] [1705921861.188818805] [wrapper.psdk_wrapper_node]: App key: **Hide**
[psdk_wrapper_node-1] [INFO] [1705921861.188825460] [wrapper.psdk_wrapper_node]: Baudrate: 921600
[psdk_wrapper_node-1] [INFO] [1705921861.188829909] [wrapper.psdk_wrapper_node]: Using connection configuration file: /home/rafa/psdk_ros2_3_8_ws/install/psdk_wrapper/share/psdk_wrapper/cfg/hw_connection.json
[psdk_wrapper_node-1] [INFO] [1705921861.188868530] [wrapper.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1705921861.188884960] [wrapper.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1705921861.188889539] [wrapper.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] Get app info failed.
[psdk_wrapper_node-1] : No such file or directory
[psdk_wrapper_node-1] Get link info failed.
[psdk_wrapper_node-1] : No such file or directory
[psdk_wrapper_node-1] [INFO] [1705921861.188923774] [wrapper.psdk_wrapper_node]: Loaded configuration file
[psdk_wrapper_node-1] [INFO] [1705921861.188928407] [wrapper.psdk_wrapper_node]: Using DJI_USE_ONLY_UART
[psdk_wrapper_node-1] [INFO] [1705921861.188932874] [wrapper.psdk_wrapper_node]: Environment has been set!
[psdk_wrapper_node-1] [INFO] [1705921861.188939681] [wrapper.psdk_wrapper_node]: Init DJI Core...
[psdk_wrapper_node-1] chmod: missing operand after ‘777’
[psdk_wrapper_node-1] Try 'chmod --help' for more information.
[psdk_wrapper_node-1] [ERROR] [1705921861.191304080] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 236. Retrying for 1 time. 
[psdk_wrapper_node-1] chmod: missing operand after ‘777’
[psdk_wrapper_node-1] Try 'chmod --help' for more information.
[psdk_wrapper_node-1] [ERROR] [1705921866.193765107] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 236. Retrying for 2 time. 
[psdk_wrapper_node-1] chmod: missing operand after ‘777’
[psdk_wrapper_node-1] Try 'chmod --help' for more information.
[psdk_wrapper_node-1] [ERROR] [1705921871.196334461] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 236. Retrying for 3 time. 
[psdk_wrapper_node-1] [WARN] [1705921876.196569679] [rcl_lifecycle]: No transition matching 3 found for current state unconfigured
[psdk_wrapper_node-1] [ERROR] [1705921876.196595133] [wrapper.psdk_wrapper_node]: Unable to start transition 3 from current state unconfigured: Transition is not registered., at ./src/rcl_lifecycle.c:355
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/wrapper/psdk_wrapper_node'
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_CONFIGURE' for LifecycleNode '/wrapper/psdk_wrapper_node'
```

Connection:
[image]

biancabnd, 2024-01-23:
In this case I would assume that you are not configuring well the psdk_ros2 wrapper. From the picture you are attaching, 
you are using the same HW configuration as in the other issue you opened, however you are setting this: DJI_USE_ONLY_UART. 
While in the other issue you were setting DJI_USE_UART_AND_USB_BULK_DEVICE. Please, make sure you are configuring well the wrapper before opening additional issues in the repository.

RPS98, 2024-01-23:
Hi @biancabnd,

I have tried both configurations, and both yield the same error. The log uses DJI_USE_ONLY_UART because I assumed it was the simplest as it has fewer things.

What could be the cause of the `Get app info failed` and `Get link info failed` errors? Could this be the source of the issue?"

biancabnd, 2024-01-25:
Hi @RPS98, 

You cannot set any HW configuration. Each one corresponds to how you are actually connected to the DJI drone. So, in your case you should use the DJI_USE_UART_AND_USB_BULK_DEVICE. Also, the configuration and settings you add here should be the same as the ones you use to run the samples. 

When this error appears: chmod: missing operand after ‘777’ what it means is that the name of the device (e.g. /dev/ttyUSB0) that you want to use to connect to the DJI drone has not been passed. Thus, what you have set in the file for some reason has not reached to the function that is using it. 

The `Get app info failed` and  `Get link info failed` errors come from the DJI Psdk code to which we only have access to the headers so there is no way to know what is really happening. But my guess is that this is due to the wrong configuration DJI_USE_ONLY_UART. Do you have the same errors when using DJI_USE_UART_AND_USB_BULK_DEVICE?

closed 2024-01-30 by vicmassy

Janphr, 2024-02-01:
Hey,
could you please simply change line 37 in `psdk_wrapper.cpp` from
`std::string default_config_file = ros_pkg_path + "/cfg/hw_connection.json";`
to
`std::string default_config_file = ros_pkg_path + "/cfg/link_config.json";`
or rename `link_config.json` to `hw_connection.json`?

biancabnd, 2024-02-01:
Hi @Janphr, it is done here https://github.com/umdlife/psdk_ros2/pull/34/commits/2e9a652c7dae293b367bc5ef4b22c083c9535ee9. But please notice this is still a feature branch (under development) and not merged yet, so some things might not be working smoothly.