umdlife/psdk_ros2 #50: how to call the sevice

lucentlc, 2024-02-04:
hi ,thanks for the great art jod.I get some question about how to call the service. I am doing the follow steps:
1.Started the psdk_wrapper and  successed.
2.I write the below code and start it,but I get  waiting the servcie.  **service not available,waiting again**
what is the problem or I miss something ?
```
import rclpy
from rclpy.node import Node
from psdk_interfaces.srv import CameraSetupStreaming

class ServiceClientRouter(Node):
    def __init__(self):
        super().__init__('service_client_router')
        self.cli=self.create_client<CameraSetupStreaming>("psdk_ros2/camera_setup_streaming")
        while not self.cli.wait_for_service(timeout_sec=1.0):
            self.get_logger().info("service not available,waiting again")
        self.req=CameraSetupStreaming.Request()

    def send_request(self,payload_index,camera_source,start_stop):
        self.req.payload_index=payload_index
        self.req.camera_source=camera_source
        self.req.start_stop=start_stop
        self.future=self.cli.call_async(self.req)
        return self.future.result()

def main():
    rclpy.init()

    service_client_router = ServiceClientRouter()
    _payload_index=7
    _camera_source=1
    _start_stop=1
    response=service_client_router.send_request(_payload_index,_camera_source,_start_stop)
    service_client_router.get_logger().info("reqeust of CameraSetupStreaming: %d,%d,%d,resp>:%d",
                                            (_payload_index,_camera_source,_start_stop,response.success))
    service_client_router.destroy_node()

    #camera_setup_streaming_service_
    rclpy.shutdown()

if __name__=="__mian__":
    main()

    

```

vicmassy, 2024-02-05:
did you make sure the topic exists by doing `ros2 service list` after spinning up the psdk wrapper?

lucentlc, 2024-02-19:
> did you make sure the service exists by doing `ros2 service list` after spinning up the psdk wrapper?

ok.Thanks for reply very much.

closed 2024-02-19 by lucentlc