umdlife/psdk_ros2 #52: Attempting to build the master branch of the repo leads to a linker error

Yamboy1, 2024-02-08:
**Tell us about your environment**

* **Ubuntu version:** Ubuntu 22.04
* **Docker Version:**
* **ROS/ROS2 version:** Humble
* **Commit hash:** c56718909f0b2f7eeebce915decc28942b5d8077

#### Steps to reproduce issue

```
Follow the getting started instructions: https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html
```

#### Expected behavior
The build complete's successfully
#### Actual behavior
The build fails on a linker error:
```
/usr/bin/ld: libpsdk_wrapper_libs.so: undefined reference to `DjiFlightController_Deinit'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/psdk_wrapper_node.dir/build.make:247: bin/psdk_wrapper_node] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/psdk_wrapper_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< psdk_wrapper [2min 25s, exited with code 2]
                                      
Summary: 1 package finished [3min 55s]
  1 package failed: psdk_wrapper
  1 package had stderr output: psdk_wrapper
```
#### Additional information
We have previous version of the repository cloned from 72cac881cc38d8086b4e5aac80f5f0bba5a2c7ee which works perfectly fine
#### Are you willing to make a PR? If yes, add the PR link here.
I'll have a look, but I'm not sure if I'm skilled enough with C++ to figure out the issue

Yamboy1, 2024-02-08:
Problem solved, we've been using v3.8 at work, and this is an incompatibility between building for 3.5 and then linking to 3.8

closed 2024-02-08 by Yamboy1

vicmassy, 2024-02-08:
Hey @Yamboy1, soon we will release a version compatible with PSDK v3.8. Stay tuned