umdlife/psdk_ros2 #63: Compilation failure of json utils

RPS98, 2024-03-01:
**Tell us about your environment**

* **Ubuntu version:** Ubuntu 22.04
* **Docker Version:** [isaac_ros_common](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common)
* **ROS/ROS2 version:** Humble
* **Commit hash:** [833c906](https://github.com/umdlife/psdk_ros2/commit/833c9064c20da18fd71d992b64845a8cb230debb)

#### Steps to reproduce issue
Compile PSDK Wrapper using [guide](https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html).

#### Expected behavior
Compile without errors.

#### Actual behavior
Output:
```
/home/admin/psdk_ws/src/psdk_ros2/psdk_wrapper/include/psdk_wrapper/json_utils
   39 |   stream << "0x" << std::setfill('0') << std::setw(sizeof(T) * 2) << s
      |             ^~~~
/home/admin/psdk_ws/src/psdk_ros2/psdk_wrapper/include/psdk_wrapper/json_utils
   39 |   stream << "0x" << std::setfill('0') << std::setw(sizeof(T) * 2) << s
      |                          ^~~~~~~
      |                          fill
/home/admin/psdk_ws/src/psdk_ros2/psdk_wrapper/include/psdk_wrapper/json_utils
   39 |   stream << "0x" << std::setfill('0') << std::setw(sizeof(T) * 2) << s
      |                                               ^~~~
/home/admin/psdk_ws/src/psdk_ros2/psdk_wrapper/include/psdk_wrapper/json_utils
   38 |   std::stringstream stream;
```

#### Are you willing to make a PR? If yes, add the PR link here.

bitcurious, 2024-03-28:
I got the same 3 error while doing colcon build  
My system is 

**Jetson AGX**

**Docker Image jetson-isaac2:latest**

bitcurious, 2024-03-28:
@RPS98 do you get the solution for this ?

bitcurious, 2024-03-29:
My issue is also resolve after adding 
**#include <sstream>
#include <iomanip>**
this packages into **json_utils.hpp** file

closed 2024-04-03 by biancabnd