umdlife/psdk_ros2 #71: Error DJI core could not be initiated

bitcurious, 2024-03-29:
**Tell us about your environment**

* **Ubuntu version:**  20.04
* **Docker Version:**24.0.5
* **ROS/ROS2 version:**Humble
* **Commit hash:**

#### Steps to reproduce issue

```
Set as per instruction
launch wrapper.launch.py file
```

#### Expected behavior
Should start the node, publisher, subscribers and services.
#### Actual behavior
Getting error initializing error is 
`[psdk_wrapper_node-1] [ERROR] [1711735486.367322347] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 225. Retrying for 0 time. `
#### Additional information
[image]

#### Are you willing to make a PR? If yes, add the PR link here.

bitcurious, 2024-04-04:
Hi @biancabnd,

I've been working with the DJI Matrice 300 (M300) drone and I mistakenly referred to it as the M30 in  to the Payload SDK (PSDK) app.

Initially, I used the M300 with the OSDK extension module and then I switched to the E-port development module with the M300 drone. However, I encountered the same issue with both setups. Interestingly, in M300 with Eport setup after power cycling the drone once, it initiated successfully and was running without any issues.

The current problem I'm facing is that I'm unable to call any service or API. When I try to run the start takeoff command, it throws the following error: **"could not start takeoff: Error code is: 124554051784"**. Additionally, I attempted to add a power management module to shutdown the drone when the battery percentage is low, but that API also didn't work as expected.

Additionally When do any small change into code I need to do power cycle of drone otherwise I shows the error **"DJI core could not be initialized"** please let me know any solution for this.
One doubt do we need to make any changes in Payload-SDK of DJI as per instructions given here[]( https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/jetson-nano.html) or we just need to git clone it and do the setup as per[]( https://umdlife.github.io/psdk_ros2/documentation/GettingStarted.html)
I've attached a screenshot of the error and the GitHub link to my complete workspace where the psdk_wrapper and Payload SDK are running for reference.

Thank you for your assistance.

Github link-  [](https://github.com/bitcurious/power_management/tree/main)

[image]

biancabnd, 2024-04-05:
Hi @bitcurious,

When using a DJI M300, we usually connect to it via the OSDK extension module. If you use it as well, then you need to make sure that  you configure properly the link_config file. The one that we give as an example should already work. You only need to modify the two uart device names uart1_device_name and uart2_device_name to your actual ports. For this connection setup, you need to use link_select = use_uart_and_usb_bulk_device.

When using the Eport, the communication is slightly different, and then you need to use link_select = use_uart_and_network_device as also specified in the DJI documentation. In this case, then yes, you might need to do some additional steps on your board to make sure that this new network interface that DJI is using to communicate to your board is indeed enabled. 

The easier setup is using the OSDK connection, then you shouldn't need to do anything. 

From the screenshots you sent me I cannot see if the psdk application is actually running correctly. Can you please attach the entire log from the beginning?

bitcurious, 2024-04-12:
@biancabnd 
The PSDK wrapper is functioning properly for telemetry and the flight control module. The issue I encountered was due to the drone being located in a No Fly Zone (NFZ), which I was unaware of while conducting remote development. After bypassing the NFZ, I am now able to control the drone.

This was tested using the OSDK extension module with the `use_uart_and_usb_bulk_device` configuration. The same configuration was also tested on the E-port development kit. I believe that for the E-port development board, using `use_only_uart` should also be effective. Despite being able to receive all data and send commands to the drone, I could not execute takeoff and flight-related commands due to the drone's location being in an NFZ.
 
 I have not attempted to integrate the camera, gimbal and liveview module yet. I have some questions regarding this integration. Do we need to specify the camera's name or model in the wrapper? or Do we only need to make it true in psdk_params.yaml file ?

I'm curious about the purpose of the `"uart2_device_name": "/dev/dji_advanced_sensing" `setting. Could you explain what type of device needs to be connected to this and what functionalities it offers?

Here is screenshot where service call get successful for take off command.

[image]

bitcurious, 2024-04-17:
Hi @biancabnd, Currently, I am using the DJI Matrice 300 (M300) drone with the Onboard SDK (OSDK) extension module and the H20T camera. However, I am facing issues with initializing the camera module when setting it to true, which prevents the wrapper from running successfully. 

Additionally, I have configured the system to use 'use_uart_and_usb_bulk_device'. 

In regards to additional information, when I ran the command `ps -aux | grep startup_bulk`, I received the following output: `root 401 0.0 0.0 9236 748 pts/6 S+ 01:53 0:00 grep --color=auto startup_bulk`. 

Furthermore, when I ran the command `lsusb`, I obtained the following output: `Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC`. 

Based on the provided VID and PID, the VID is `0403` and the PID is `6001` is used in usb_bulk_config. 

Please advise on the required configuration to successfully obtain the camera stream.

[image]

biancabnd, 2024-04-19:
@bitcurious  based on your comment on other issues, can we close this one now?

bitcurious, 2024-04-23:
Yes we can close this. Thank you so much @biancabnd for your support.

closed 2024-04-23 by biancabnd