umdlife/psdk_ros2 #97: Perception module integration for Stereo Vision Image/feed

bitcurious, 2024-05-21:
## Feature request

#### Feature description

The visual perception system of the DJI's drone is composed of the visual perception sensor and visual perception algorithm. During the flight of the drone, the perception sensor can obtain the state of the surrounding environment, assist drone to brake, avoid obstacle and accuracy hover.
Currently only relative obstacle data is provided by this wrapper and not the Stereo Vision.
#### Implementation considerations

[DJI API link for Perception](https://developer.dji.com/doc/payload-sdk-api-reference/en/practice/perception.html
) APIs and fucntions mentioned in this link can be integrared 

Pros:

1. Help in auonomous perspective navigation
2. More accureccy in localization

#### Are you willing to submit a pull request to implement this change?

biancabnd, 2024-05-23:
Hi @bitcurious, 
Thanks for the suggestion. It is true that this module is still not implemented within the wrapper. Whenever we will have the availability we will add it. Meanwhile, feel free to open a PR and implement it. 
Thanks again.

bitcurious, 2024-05-23:
@biancabnd I'm trying. I will let you know if i need any help.

biancabnd, 2024-05-23:
Sure, let me know. I don't have the availability to develop it myself but I can support.

umesh-mane, 2024-06-10:
@biancabnd  I have successfully added and tested the perception module in this wrapper. However, the wrapper has been updated to a structure where each module has a different node and thread. I will update my code accordingly and raise a pull request for it within the next 2-3 days.

bitcurious, 2024-06-10:
@biancabnd  I have successfully added and tested the perception module in this wrapper. However, the wrapper has been updated to a structure where each module has a different node and thread. I will update my code accordingly and raise a pull request for it within the next 2-3 days.