umdlife/psdk_ros2 #99: No response from the drone when sending velocity commands

Aspirinkb, 2024-05-23:
Issue: No response from the drone when sending velocity commands via topic /psdk_ros2/flight_control_setpoint_FLUvelocity_yawrate, even when manually controlling the drone's ascent using a remote controller (in the simulator).

Description:
When attempting to send velocity commands to the drone using the topic /psdk_ros2/flight_control_setpoint_FLUvelocity_yawrate, there is no response or action observed from the unmanned aerial vehicle (UAV) in the simulator.

cmd:
`ros2 topic pub --times 100 --rate 20 /wrapper/psdk_ros2/flight_control_setpoint_FLUvelocity_yawrate sensor_msgs/msg/Joy "{axes: [0,0,2,0]}"`

vicmassy, 2024-05-23:
did you first request authority to control the drone autonomously?

vicmassy, 2024-05-23:
yes! Please ensure you get familiarized with the PSDK documentation from DJI as well as some sample codes in C++.

closed 2024-05-23 by vicmassy

Aspirinkb, 2024-05-24:
I create a ROS2 package which has a node to obtain authority, the code as following:
```
  void obtain_authority_callback(const std_msgs::msg::String & message) //const
  {
    RCLCPP_INFO(this->get_logger(), "I heard obtaining UAV authority command: %s", message.data.c_str());
    T_DjiOsalHandler *s_osalHandler = DjiPlatform_GetOsalHandler();
    T_DjiFlightControllerRidInfo ridInfo = {22.542812, 113.958902, 0};
    T_DjiReturnCode returnCode;
    returnCode = DjiFlightController_Init(ridInfo);
    if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
      // USER_LOG_ERROR("Init flight Control sample failed,error code:0x%08llX", returnCode);
      RCLCPP_INFO(this->get_logger(), "Failed to initialize flight control");
    }
    returnCode = DjiFlightController_ObtainJoystickCtrlAuthority();
    if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        // USER_LOG_ERROR("Obtain joystick authority failed, error code: 0x%08X", returnCode);
        RCLCPP_INFO(this->get_logger(), "Failed to obtain uav authority");
    }
    if (returnCode == DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
      RCLCPP_INFO(this->get_logger(), "Successfully obtain UAV authority! Now psdk can control flight.");
    }
    s_osalHandler->TaskSleepMs(1000);
  }
```

But, if I run psdk_ros2 package first and the obtain_authority node can not run. Maybe it is about serial device? 
So, how to run psdk_ros2 and other nodes those using DJI serial device at the same time?

vicmassy, 2024-05-24:
we offer a service that does that! That is the whole point of the psdk_ros2 wrapper! You do not need to communicate directly with the PSDK or any serial communication

Aspirinkb, 2024-05-24:
> we offer a service that does that! That is the whole point of the psdk_ros2 wrapper! You do not need to communicate directly with the PSDK or any serial communication

Great! `/wrapper/psdk_ros2/obtain_ctrl_authority`~