umdlife/psdk_ros2 #103: Struggeling with the installation

BallerNacken, 2024-05-24:
Hi, 

I am a real newbie in this field. I would like to use the ros2 wrapper for my M300 to obtain the drones data for later processes. I was able to build the payload sdk and connect it to the M300. Now I would like to use ROS2 to actually get the data ans store it .csv files somehow (not even sure if ROS2 supports this). 

But I am already struggeling in installing/building ROS2. The documentation is not very helpful for me as I am not really understanding what for example is meant by putting the psdk and ros2 into the same workspace? Does PSDK have to be compiled already?
Right now it is simply compiled/build in my home directory. Can someone elaborate in more detail on how I am installing/building the ROS2 wrapper?

biancabnd, 2024-05-27:
Hi @BallerNacken, 
The `psdk_ros2` wrapper is based on ROS 2 Humble, so you should first make sure you have a clean ROS 2 Humble installation. I suggest you to read their documentation and get familiarized with it: https://docs.ros.org/en/humble/index.html. 
After that, you can opt for the simple debian installation of both psdk wrapper and psdk interfaces as specified in our documentation https://umdlife.github.io/psdk_ros2/documentation/RunningTheWrapper.html#running-the-wrapper. With this type of installation, you don not need to compile anything. The only thing remaining to do is to properly configure the link_config and params files.

BallerNacken, 2024-05-28:
Hi @biancabnd,
Thanks for the explanation. I was able to install everything including the psdk_wrapper! But running the psdk_wrapper I end up in some errors:

`[INFO] [launch]: All log files can be found below /home/m300/.ros/log/2024-05-28-09-53-18-112979-m300-2998
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [psdk_wrapper_node-1]: process started with pid [3009]
[psdk_wrapper_node-1] [INFO] [1716882798.664129929] [wrapper.psdk_wrapper_node]: Creating Constructor PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1716882798.747353053] [wrapper.psdk_wrapper_node]: Configuring PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1716882798.747586755] [wrapper.psdk_wrapper_node]: Loading parameters
[psdk_wrapper_node-1] [INFO] [1716882798.747785826] [wrapper.psdk_wrapper_node]: App name: xxx
[psdk_wrapper_node-1] [INFO] [1716882798.747904214] [wrapper.psdk_wrapper_node]: App id: xxx
[psdk_wrapper_node-1] [INFO] [1716882798.748015750] [wrapper.psdk_wrapper_node]: App key: xxx
[psdk_wrapper_node-1] [INFO] [1716882798.748130897] [wrapper.psdk_wrapper_node]: Baudrate: 921600
[psdk_wrapper_node-1] [INFO] [1716882798.748233747] [wrapper.psdk_wrapper_node]: Using connection configuration file: /opt/ros/humble/share/psdk_wrapper/cfg/link_config.json
[psdk_wrapper_node-1] [INFO] [1716882798.748474430] [wrapper.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1716882798.748739316] [wrapper.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1716882798.748882777] [wrapper.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] [INFO] [1716882798.754212367] [wrapper.psdk_wrapper_node]: Loaded configuration file
[psdk_wrapper_node-1] [INFO] [1716882798.754665455] [wrapper.psdk_wrapper_node]: Using DJI_USE_ONLY_UART
[psdk_wrapper_node-1] [INFO] [1716882798.754805342] [wrapper.psdk_wrapper_node]: Environment has been set!
[psdk_wrapper_node-1] [INFO] [1716882798.756361788] [wrapper.psdk_wrapper_node]: Activating PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1716882798.756588971] [wrapper.psdk_wrapper_node]: Init DJI Core...
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/dji_serial': Operation not permitted
[psdk_wrapper_node-1] sh: 1: cannot create /proc/sys/net/core/rmem_default: Permission denied
[psdk_wrapper_node-1] sh: 1: cannot create /proc/sys/net/core/rmem_max: Permission denied
[psdk_wrapper_node-1] [ERROR] [1716882804.007220972] [wrapper.psdk_wrapper_node]: Please use the extension port
[psdk_wrapper_node-1] [ERROR] [1716882804.012012752] [wrapper.psdk_wrapper_node]: Failed to finish transition 3. Current state is now: inactive (Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368)
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/wrapper/psdk_wrapper_node'
[psdk_wrapper_node-1] [ERROR] [1716882804.130901374] [wrapper.psdk_wrapper_node]: Unable to start transition 6 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[psdk_wrapper_node-1] [WARN] [1716882804.131051780] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[ERROR] [psdk_wrapper_node-1]: process has died [pid 3009, exit code -11, cmd '/opt/ros/humble/lib/psdk_wrapper/psdk_wrapper_node --ros-args -r __node:=psdk_wrapper_node -r __ns:=/wrapper --params-file /tmp/launch_params_fznage2t --params-file /opt/ros/humble/share/psdk_wrapper/cfg/psdk_params.yaml'].`

I set the UDEV rule according to the documentation, but somewhere in the script it is trying to change the permissions. Not sure what is going on here. And also later it tries to create files/folders but has not the permissions. So there seems to be a permission error somewhere?

Info: Trying to connect to a M300 using the Skyport V2 connected to a PSDK port (OSDK is not an option as this needs to be used for a parachute).

biancabnd, 2024-05-29:
Hi @BallerNacken, 
If you are using the Skyport V2, the link configuration of the psdk wrapper will be different than the default one and you don't necessarily need to use the udev rules. We have not tested it using this type of connection, but as you can see in DJI's documentation (https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/device-connect.html) this type of connection uses a serial and network port to transmit the data. Thus, I think you need to configure the `link_available` parameter as `use_uart_and_network_device`. 
In any case, first of all you should make sure that you are able to run the sample code from DJI using your connection. If you can do that, then the setup you have used to run the sample code, should be similar to the one you need to configure for the psdk wrapper.

bonchan, 2024-06-17:
@BallerNacken can you show your udev rule? I had the same Issue.
The solution was to avoid the symlink. 
changed link_config.json to this
`        "uart_config": {
            "uart1_device_name": "/dev/ttyUSB0",
            "uart2_device_enable": "true",
            "uart2_device_name": "/dev/ttACM0"
        },`
 and the udev rule looks like this:
`SUBSYSTEM=="usb", ATTRS{idVendor}=="2ca3", MODE="0666"`

hope it helps