umdlife/psdk_ros2 #107: Failed to run as non root users

Aspirinkb, 2024-05-29:
I only can run the psdk-ros2 as root user.    

Run it as non root user, errors as following:
```
[INFO] [launch]: All log files can be found below /home/zhy/.ros/log/2024-05-29-11-37-51-693863-ubuntu-66239
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [psdk_wrapper_node-1]: process started with pid [66250]
[psdk_wrapper_node-1] [INFO] [1716953871.938300117] [wrapper.psdk_wrapper_node]: Creating Constructor PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1716953872.405925383] [wrapper.psdk_wrapper_node]: Configuring PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1716953872.406324552] [wrapper.psdk_wrapper_node]: Loading parameters
[psdk_wrapper_node-1] [INFO] [1716953872.406767529] [wrapper.psdk_wrapper_node]: App name: test
[psdk_wrapper_node-1] [INFO] [1716953872.406815049] [wrapper.psdk_wrapper_node]: App id: 146265
[psdk_wrapper_node-1] [INFO] [1716953872.406850249] [wrapper.psdk_wrapper_node]: App key: fe587302b376d8a81aaca870e3db66b
[psdk_wrapper_node-1] [INFO] [1716953872.406901193] [wrapper.psdk_wrapper_node]: Baudrate: 460800
[psdk_wrapper_node-1] [INFO] [1716953872.406937801] [wrapper.psdk_wrapper_node]: Using connection configuration file: /home/zhy/psdk_ros2_ws/src/psdk_ros2/psdk_wrapper/cfg/link_config.json
[psdk_wrapper_node-1] [INFO] [1716953872.407319817] [wrapper.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1716953872.407509514] [wrapper.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1716953872.407556554] [wrapper.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] [INFO] [1716953872.407581066] [wrapper.psdk_wrapper_node]: Loaded configuration file
[psdk_wrapper_node-1] [INFO] [1716953872.407597290] [wrapper.psdk_wrapper_node]: Using DJI_USE_UART_AND_NETWORK_DEVICE
[psdk_wrapper_node-1] [INFO] [1716953872.407650154] [wrapper.psdk_wrapper_node]: Environment has been set!
[psdk_wrapper_node-1] [INFO] [1716953872.408285419] [wrapper.psdk_wrapper_node]: Activating PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1716953872.408383019] [wrapper.psdk_wrapper_node]: Init DJI Core...
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyTHS0': Operation not permitted
...
[psdk_wrapper_node-1] [ERROR] [1716953900.394957751] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 225. Retrying for 1 time.
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyTHS0': Operation not permitted
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyTHS0': Operation not permitted
```
Could you help to run it as non root user?

biancabnd, 2024-05-29:
Hi @Aspirinkb, 
These requirements are not from the wrapper itself but from the Payload-SDK libraries. To run it as non root you can try to add your user to the dialout group or create an udev rule for this device which grants you the needed permissions. 
Hope it helps!

umesh-mane, 2024-06-10:
@biancabnd I attempted to run the program as a regular user by adding the user to the dialout group and creating a udev rule. However, it throws the following error. When I run the program as root, it executes successfully without any errors. I need to determine if the error is due to a lack of access to the serial port or an inability to access the RCLCPP Lifecycle node publisher.
>  [psdk_wrapper_node-1] [ERROR] [1718023100.202970640] [wrapper.camera_node]: Could not initialize camera manager. Error code: 227
> [psdk_wrapper_node-1] [ERROR] [1718023100.211768970] [wrapper.psdk_wrapper_node]: Failed to finish transition 3. Current state is now: inactive (Could not publish transition: publisher's context is invalid, at /tmp/binarydeb/ros-humble-rcl-5.3.6/src/rcl/publisher.c:389, at /tmp/binarydeb/ros-humble-rcl-lifecycle-5.3.6/src/rcl_lifecycle.c:368)
> [ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/wrapper/psdk_wrapper_node'
> [psdk_wrapper_node-1] [INFO] [1718023100.212966214] [wrapper.psdk_wrapper_node]: Destroying PSDKWrapper

[image]

bonchan, 2024-06-17:
@umesh-mane did you manage to solve this permission issues?

umesh-mane, 2024-06-17:
@bonchan  No I'm onle able to get the permission for /dev/ttyUSB0 port and not for /dev/ttyACM0. resulting I'm able to get only telemetry and flight control module and not the camera module.

vicmassy, 2024-06-17:
Hey @bonchan AND @umesh-mane, 
try as suggested by @biancabnd, this 
[image]

It is referenced here https://developer.dji.com/onboard-sdk/documentation/development-workflow/environment-setup.html 

Even though it says its for the OSDK, the permisions needed are the same

bonchan, 2024-06-17:
@vicmassy that did it. thanks
I had this rules set for ttyUSB0 and ttyACM0
lines 1 to 6
[image]
but i had them as symlinks. maybe that was my problem

I changed the file and only left lines 9 and 10

when i start psdk_ros2 it still shows that psdk lib(dji) is trying to change the permissions but video stream works fine,. Ill try to use the symlinks and get back here.

umesh-mane, 2024-06-18:
@bonchan @vicmassy it works for me as well without symlink.
@biancabnd which drone are you using ?

bonchan, 2024-06-18:
@umesh-mane M300 + Jetson Orin Nano
Im still having some issues with the bulk configuration but for now im ok to go forward

umesh-mane, 2024-06-20:
@bonchan Okay. 
have tried to setup bulk setup ? [USB bulk setup](https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/jetson-nano.html#_2-enable-jetson-nano-s-usb-bulk-function) you can check it here.
You are using E-port development kit right ?

bonchan, 2024-06-20:
@umesh-mane yes, I'm using the E-port. and yes. I have tried to setup bulk configuration following that guide. I managed to do it on jetson nano but now I switched to jetson orin nano, I'll get back here once I make it work.

umesh-mane, 2024-06-28:
@vicmassy @biancabnd 
`[psdk_wrapper_node-14] chmod: changing permissions of '/dev/ttyUSB0': Operation not permitted
[psdk_wrapper_node-14] chmod: changing permissions of '/dev/ttyACM0': Operation not permitted`
Is this permission change happening because of the psdk_wrapper code, or is it occurring from the DJI_PSDK side?

biancabnd, 2024-06-28:
Hi. This is required by the DJI PSDK not the wrapper itself.

umesh-mane, 2024-06-28:
@biancabnd Okay
Thanks for update.

Aspirinkb, 2024-07-02:
I am using `DJI_USE_UART_AND_NETWORK_DEVICE` configuration to transform serial port data (UART) and camera data (Network). 
I have add my user to `dialout` group and relogin.    
I have encountered the same issue (**Operation not permitted**) persisting without any changes. I would appreciate guidance on how to proceed further.
**Operation not permitted**  tip:   
```
[psdk_wrapper_node-1] chmod: changing permissions of '/dev/ttyTHS0': Operation not permitted
```   
**check serial port device and user group**:
```
# check serial port device ttyTHS0, user group
 ls /dev/ttyTHS0 -la
crwxrwxrwx 1 root dialout 240, 0  7月  2 15:27 /dev/ttyTHS0

# check dialout group members
grep dialout /etc/group
dialout:x:20:<my_user>
```

Aspirinkb, 2024-07-02:
> @umesh-mane yes, I'm using the E-port. and yes. I have tried to setup bulk configuration following that guide. I managed to do it on jetson nano but now I switched to jetson orin nano, I'll get back here once I make it work.

@bonchan I give up using usb bulk to transform camera data on Jetson Orin, using network is ok.

biancabnd, 2024-07-02:
> I am using `DJI_USE_UART_AND_NETWORK_DEVICE` configuration to transform serial port data (UART) and camera data (Network). I have add my user to `dialout` group and relogin. I have encountered the [...]

Hi @Aspirinkb, 

To solve this issue you could change the permissions of the specific device `sudo chmod 777 /dev/ttyTHS0`. If you want this change to be persistent, you can create an udev rule which grants the needed permissions to this specific device. In case you are using a docker to launch the psdk wrapper, you could run the docker with the privileged flag set to true. 

Hope it helps.

Aspirinkb, 2024-07-02:
> Hi @Aspirinkb,
>
> To solve this issue you could change the permissions of the specific device `sudo chmod 777 /dev/ttyTHS0`. If you want this change to be persistent, you can create an udev rule which [...]

Thanks @biancabnd, but I did change the permissions of the specific device with `sudo chmod 777 /dev/ttyTHS0`, as you can see above when check device ttyTHS0 `crwxrwxrwx 1 root dialout 240, 0  7月  2 15:27 /dev/ttyTHS0`

biancabnd, 2024-07-02:
I see, this should already work. How are you running the wrapper?

whyscience, 2024-08-01:
> Execute the C++ language sample program
Go to the directory of the sample program: cd build/bin/
Use the sudo ./dji_sdk_demo_linux_cxx command to run the sample program

DJI expects we to use sudo, so they do not add sudo in the code 
`sprintf(systemCmd, "chmod 777 %s", uartName);`

it should be `sprintf(systemCmd, "sudo chmod 777 %s", uartName);`

But ros2 launch is usually non sudo user.

So we can do the following: 
1. change to sudo without password: https://gcore.com/learning/how-to-disable-password-for-sudo-command/
2. add `sudo` to such code 

Refer to my branch : https://github.com/whyscience/Payload-SDK/tree/master

bitcurious, 2024-08-16:
@whyscience Were you able to run the PSDK wrapper without root permissions after making changes to the DJI PSDK repository's sample code?

whyscience, 2024-08-19:
@bitcurious of course, you need to make sure sudo do not need password

bitcurious, 2024-08-21:
@whyscience I tried the method you mentioned above. I made changes to the sudoers file and also referred to your commit, updating the code accordingly, but it still doesn't work for me. Here's what my sudoers file looks like.

[image]

whyscience, 2024-08-21:
You still need to make some changes to the init code of the ros repo if you are using EPort

获取 Outlook for iOS<https://aka.ms/o0ukef>

image.png (view on web)<https://github.com/user-attachments/assets/a87b80cd-e2f7-419d-adf6-83a19712be6f>

bitcurious, 2024-08-21:
No I'm not using E-port. I'm using OSDK Extension module with M300 drone.

whyscience, 2024-08-21:
Same thing, you need copy full init code from dji cpp sample to the ros init part

获取 Outlook for iOS<https://aka.ms/o0ukef>

bitcurious, 2024-08-21:
okay

bitcurious, 2024-08-21:
Do you have any reference for it ?
if possible could you please share it ?

whyscience, 2024-08-21:
bool
PSDKWrapper::set_environment()
{
  RCLCPP_INFO(get_logger(), "Setting environment");
  T_DjiReturnCode return_code;
  T_DjiOsalHandler osal_handler = {0};
  T_DjiHalUartHandler uart_handler = {0};

  T_DjiHalUsbBulkHandler usb_bulk_handler = {0};
  T_DjiLoggerConsole print_console;
  T_DjiLoggerConsole local_record_console;
  T_DjiFileSystemHandler file_system_handler = {0};
  T_DjiSocketHandler socket_handler{0};
  T_DjiHalNetworkHandler network_handler = {0};
  T_DjiUserLinkConfig linkConfig;
  network_handler.NetworkInit = HalNetWork_Init;
  network_handler.NetworkDeInit = HalNetWork_DeInit;
  network_handler.NetworkGetDeviceInfo = HalNetWork_GetDeviceInfo;
  socket_handler.Socket = Osal_Socket;
  socket_handler.Bind = Osal_Bind;
  socket_handler.Close = Osal_Close;
  socket_handler.UdpSendData = Osal_UdpSendData;
  socket_handler.UdpRecvData = Osal_UdpRecvData;
  socket_handler.TcpListen = Osal_TcpListen;
  socket_handler.TcpAccept = Osal_TcpAccept;
  socket_handler.TcpConnect = Osal_TcpConnect;
  socket_handler.TcpSendData = Osal_TcpSendData;
  socket_handler.TcpRecvData = Osal_TcpRecvData;
  osal_handler.TaskCreate = Osal_TaskCreate;
  osal_handler.TaskDestroy = Osal_TaskDestroy;
  osal_handler.TaskSleepMs = Osal_TaskSleepMs;
  osal_handler.MutexCreate = Osal_MutexCreate;
  osal_handler.MutexDestroy = Osal_MutexDestroy;
  osal_handler.MutexLock = Osal_MutexLock;
  osal_handler.MutexUnlock = Osal_MutexUnlock;
  osal_handler.SemaphoreCreate = Osal_SemaphoreCreate;
  osal_handler.SemaphoreDestroy = Osal_SemaphoreDestroy;
  osal_handler.SemaphoreWait = Osal_SemaphoreWait;
  osal_handler.SemaphoreTimedWait = Osal_SemaphoreTimedWait;
  osal_handler.SemaphorePost = Osal_SemaphorePost;
  osal_handler.Malloc = Osal_Malloc;
  osal_handler.Free = Osal_Free;
  osal_handler.GetTimeMs = Osal_GetTimeMs;
  osal_handler.GetTimeUs = Osal_GetTimeUs;
  osal_handler.GetRandomNum = Osal_GetRandomNum;
  print_console.func = DjiUser_PrintConsole;
  print_console.consoleLevel = DJI_LOGGER_CONSOLE_LOG_LEVEL_INFO;
  print_console.isSupportColor = true;
  local_record_console.consoleLevel = DJI_LOGGER_CONSOLE_LOG_LEVEL_DEBUG;
  local_record_console.func = DjiUser_LocalWrite;
  local_record_console.isSupportColor = false;
  uart_handler.UartInit = HalUart_Init;
  uart_handler.UartDeInit = HalUart_DeInit;
  uart_handler.UartWriteData = HalUart_WriteData;
  uart_handler.UartReadData = HalUart_ReadData;
  uart_handler.UartGetStatus = HalUart_GetStatus;
  usb_bulk_handler.UsbBulkInit = HalUsbBulk_Init;
  usb_bulk_handler.UsbBulkDeInit = HalUsbBulk_DeInit;
  usb_bulk_handler.UsbBulkWriteData = HalUsbBulk_WriteData;
  usb_bulk_handler.UsbBulkReadData = HalUsbBulk_ReadData;
  usb_bulk_handler.UsbBulkGetDeviceInfo = HalUsbBulk_GetDeviceInfo;
  file_system_handler.FileOpen = Osal_FileOpen;
  file_system_handler.FileClose = Osal_FileClose;
  file_system_handler.FileWrite = Osal_FileWrite;
  file_system_handler.FileRead = Osal_FileRead;
  file_system_handler.FileSync = Osal_FileSync;
  file_system_handler.FileSeek = Osal_FileSeek;
  file_system_handler.DirOpen = Osal_DirOpen;
  file_system_handler.DirClose = Osal_DirClose;
  file_system_handler.DirRead = Osal_DirRead;
[... 72 lines cut ...]
  {
    RCLCPP_ERROR(get_logger(),
                 "Register OSAL filesystem handler error.Error code is: %ld",
                 return_code);
    return false;
  }
  if (DjiUser_LocalWriteFsInit("Logs/DJI") != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
    RCLCPP_ERROR(get_logger(), "File system init error.");
    return false;
  }
  return_code = DjiLogger_AddConsole(&print_console);
  if (return_code != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
    RCLCPP_ERROR(get_logger(), "Add printf console error.");
    return false;
  }
  return_code = DjiLogger_AddConsole(&local_record_console);
  if (return_code != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
    RCLCPP_ERROR(get_logger(), "Add printf console error.");
    return false;
  }
  RCLCPP_INFO(get_logger(), "Environment has been set!");
  return true;
}

获取 Outlook for iOS<https://aka.ms/o0ukef>

bitcurious, 2024-08-22:
@whyscience I'm getting following error while building the package
`/home/ubuntu/psdk_ws/src/psdk_ros2/psdk_wrapper/src/psdk_wrapper.cpp:302:24: error: 'DjiUser_PrintConsole' was not declared in this scope
  302 |   print_console.func = DjiUser_PrintConsole;
      |                        ^~~~~~~~~~~~~~~~~~~~
/home/ubuntu/psdk_ws/src/psdk_ros2/psdk_wrapper/src/psdk_wrapper.cpp:306:31: error: 'DjiUser_LocalWrite' was not declared in this scope
  306 |   local_record_console.func = DjiUser_LocalWrite;
      |                               ^~~~~~~~~~~~~~~~~~
/home/ubuntu/psdk_ws/src/psdk_ros2/psdk_wrapper/src/psdk_wrapper.cpp:405:7: error: 'DjiUser_LocalWriteFsInit' was not declared in this scope
  405 |   if (DjiUser_LocalWriteFsInit("Logs/DJI") != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/psdk_wrapper_libs.dir/build.make:76: CMakeFiles/psdk_wrapper_libs.dir/src/psdk_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/psdk_wrapper_libs.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< psdk_wrapper [27.0s, exited with code 2]`

Can you let me know how can I declear this ?

whyscience, 2024-08-22:
Copy these function from dji sample

获取 Outlook for iOS<https://aka.ms/o0ukef>

bitcurious, 2024-08-22:
in psdk_wraper.cpp code only ?

whyscience, 2024-08-22:
Refer to the sample, copy their code

获取 Outlook for iOS<https://aka.ms/o0ukef>

bitcurious, 2024-08-22:
@whyscience 
```
/home/ubuntu/psdk_ws/src/psdk_ros2/psdk_wrapper/src/psdk_wrapper.cpp:270:46: warning: missing initializer for member ‘T_DjiHalNetworkHandler::NetworkGetDeviceInfo’ [-Wmissing-field-initializers]
/home/ubuntu/psdk_ws/src/psdk_ros2/psdk_wrapper/src/psdk_wrapper.cpp:302:24: error: cannot convert ‘psdk_ros2::PSDKWrapper::DjiUser_PrintConsole’ from type ‘T_DjiReturnCode (psdk_ros2::PSDKWrapper::)(const uint8_t*, uint16_t)’ {aka ‘long unsigned int (psdk_ros2::PSDKWrapper::)(const unsigned char*, short unsigned int)’} to type ‘ConsoleFunc’ {aka ‘long unsigned int (*)(const unsigned char*, short unsigned int)’}
  302 |   print_console.func = DjiUser_PrintConsole;
      |                        ^~~~~~~~~~~~~~~~~~~~
/home/ubuntu/psdk_ws/src/psdk_ros2/psdk_wrapper/src/psdk_wrapper.cpp:306:31: error: cannot convert ‘psdk_ros2::PSDKWrapper::DjiUser_LocalWrite’ from type ‘T_DjiReturnCode (psdk_ros2::PSDKWrapper::)(const uint8_t*, uint16_t)’ {aka ‘long unsigned int (psdk_ros2::PSDKWrapper::)(const unsigned char*, short unsigned int)’} to type ‘ConsoleFunc’ {aka ‘long unsigned int (*)(const unsigned char*, short unsigned int)’}
  306 |   local_record_console.func = DjiUser_LocalWrite;
      |                               ^~~~~~~~~~~~~~~~~~
```

I copied that code, but DjiUser_PrintConsole is defined to return a value of type T_DjiReturnCode, while print_console is defined as T_DjiLoggerConsole. The same issue occurs with DjiUser_LocalWrite, causing an error during code compilation.

bitcurious, 2024-08-22:
@whyscience, I've updated the code in psdk_ros2_wrapper based on your suggestions and cloned the Payload-SDK official repository, making changes according to your commit. I also updated the sudoers file as instructed. However, when I run the ROS 2 psdk_ros2 wrapper launch file, it attempts to change permissions and then initializes all modules after some time.

Could you help identify the possible issues? Additionally, I'm concerned that since we're fetching data in the CMakeLists.txt file from GitHub using a link and git tag, it might be causing issues because it's retrieving data from there rather than my local repository.

whyscience, 2024-08-22:
You can comment those lines and refer to your local repo

获取 Outlook for iOS<https://aka.ms/o0ukef>

bitcurious, 2024-08-27:
@whyscience Thanks for instructions. I have configured it successfully and now able to run the wrapper without root user and password.

bitcurious, 2024-08-27:
@biancabnd @vicmassy @whyscience @bonchan @Aspirinkb 
I've successfully configured the code to run as a non-root user without requiring a password. I'm considering creating a separate branch in the psdk_ros2 wrapper, named non-root-user, where we can include this configuration and provide instructions for running the PSDK wrapper as a non-root user. Let me know your thoughts on this.

whyscience, 2024-08-27:
Cool, waiting to see it. @bitcurious

biancabnd, 2024-08-27:
Sounds great!