umdlife/psdk_ros2 #112: Dual stream H20 H20T.

bonchan, 2024-06-24:
I have replicated the stream functionality with the second payload. but I get a lot of lag. I'm not good at C++ and I don't know if what I'm doing is correct. Would this be possible or the drone does not have the processing power to do it?
I'm using an M300 with a Jetson Orin Nano

this is the branch with the changes 
https://github.com/umdlife/psdk_ros2/compare/main...bonchan:psdk_ros2:feature/multi_stream

https://youtube.com/shorts/Y4zS266a-qk

vicmassy, 2024-06-25:
I assume you want to stream both cameras at the same time. Id that is the case I do not know if DJI allows it given that the bandwidth of the transmission is limited using the USB from the drone to an embedded computer. It is something we have never tried yet but we are very keen on getting some data from this. Would you care to describe more specifically what is your intention, setup and results?

bonchan, 2024-06-26:
@vicmassy sorry for the late reply, the intention is to process 2 camera feeds at the same time. we have the dual mount with H20 and H20T cameras. 
the project is to combine the 2 camera streams and sync passing cars. 
I oppened the question to see if someone is trying to do the same.
I'll investigate a bit more why is it that I'm getting all that lag and get back here

closed 2025-01-29 by bonchan