umdlife/psdk_ros2 #113: Issue with Disabling Modules

umesh-mane, 2024-06-28:
@biancabnd @vicmassy I have set the HMS module to false, but it still gets initialized. I do not want the HMS module to be initialized. Can you fix this issue?
I tried to chnage the values in psdk_wrapper.hpp file still it initiated.

biancabnd, 2024-06-28:
Hi @umesh-mane,
The mandatory module boolean simply means that if the module is set to mandatory and it fails to initialize  an error will be issued. If a module is not mandatory, it will still initialize, but if the initialization fails, then the wrapper will only issue a warning and will continue running. So, it does not mean that the wrapper will not be initialized. 

In which way the initialization of the HMS module affects your system? Or why would you need to stop it?

umesh-mane, 2024-07-01:
It was contineously sending warining msg while development thats why i wanted to stop it.

umesh-mane, 2024-07-03:
@biancabnd we can close this issue.

closed 2024-07-03 by biancabnd