umdlife/psdk_ros2 #120: Continuous Log Issue After Starting Main Camera Stream

bitcurious, 2024-07-18:
After starting the main camera stream on the M300 drone with an H20T camera and OSDK extension module, I am receiving the following log messages continuously:

```
[psdk_wrapper_node-1] [INFO] [1721287407.390769526] [wrapper.liveview_node]: Starting streaming...
[psdk_wrapper_node-1] [INFO] [1721287407.899753403] [wrapper.liveview_node]: Successfully started the camera streaming.
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] non-existing PPS 0 referenced
[psdk_wrapper_node-1] [h264 @ 0xffff44058cf0] non-existing PPS 0 referenced
[psdk_wrapper_node-1] [h264 @ 0xffff44058cf0] decode_slice_header error
[psdk_wrapper_node-1] [h264 @ 0xffff44058cf0] no frame!
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] SEI type 1 size 1040 truncated at 848
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] non-existing PPS 0 referenced
[psdk_wrapper_node-1] [h264 @ 0xffff440206d0] SEI type 1 size 1040 truncated at 832
[psdk_wrapper_node-1] [h264 @ 0xffff440206d0] non-existing PPS 0 referenced
[psdk_wrapper_node-1] [h264 @ 0xffff440206d0] decode_slice_header error
[psdk_wrapper_node-1] [h264 @ 0xffff44058cf0] Missing reference picture, default is 0
[psdk_wrapper_node-1] [h264 @ 0xffff44058cf0] decode_slice_header error
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] SEI type 1 size 1040 truncated at 848
[psdk_wrapper_node-1] [h264 @ 0xffff440206d0] SEI type 1 size 1040 truncated at 832
[psdk_wrapper_node-1] [swscaler @ 0xffff1405b600] No accelerated colorspace conversion found from yuv420p to rgb24.
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] SEI type 1 size 1040 truncated at 848
[psdk_wrapper_node-1] [h264 @ 0xffff44058cf0] SEI type 1 size 1040 truncated at 832
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] SEI type 1 size 1040 truncated at 848
[psdk_wrapper_node-1] [h264 @ 0xffff440206d0] SEI type 1 size 1040 truncated at 832
[psdk_wrapper_node-1] [h264 @ 0xffff4403d690] SEI type 1 size 1048 truncated at 848
[psdk_wrapper_node-1] [h264 @ 0xffff440579f0] SEI type 1 size 1048 truncated at 832
```

These logs are cluttering the terminal at a very high rate. How can I remove or suppress these log messages?

vicmassy, 2024-07-18:
Hey, this comes from the DJI decoder code. We will do a follow up fix to silent those messages. 
Thank you

bitcurious, 2024-07-18:
@vicmassy Thank you for update.