umdlife/psdk_ros2 #121: Some Doubts Regarding Flight Controller Operation in /psdk_ros2/flight_control_setpoint_rollpitch_yawrate_thrust and Downloading Picture Action

bitcurious, 2024-07-19:
When I test flight operations using `/psdk_ros2/flight_control_setpoint_rollpitch_yawrate_thrust` (which I believe is attitude control), I have to set the `z_cmd` value to more than 35.0 to keep the drone at the respective altitude. If I set it to less than 35, the drone descends continuously amd if i set more than 35, then drone ascends contineously. 

Please let me know how we can control the drone using the `/psdk_ros2/flight_control_setpoint_rollpitch_yawrate_thrust` topic.

For downloading an image, I'm using the following action through the terminal:
```bash
ros2 action send_goal /wrapper/psdk_ros2/camera_download_file_by_index psdk_interfaces/action/CameraDownloadFileByIndex "{payload_index: 1, file_index: 1, file_name: 'dji_pic.jpeg', file_path: '/home/ubuntu/'}"
```
However, it's not successful. Can you check if I'm calling it correctly or if there's an issue?

biancabnd, 2024-07-23:
Hi @bitcurious ,
Is the file index the one you retrieve from the other action call?

bitcurious, 2024-08-13:
@biancabnd, Sorry for the late reply; I didn't have access to the camera for a while. The problem was due to not specifying the file index correctly. I resolved it by using the `/wrapper/psdk_ros2/camera_get_file_list_info` service to retrieve all the file information, including the file index. 
`success=True, file_list=[psdk_interfaces.msg.FileInfo(name='DJI_20240517103721_0001_T.JPG', type=0, size=871888, index=262161, create_time_unix=1715971040, attributes=psdk_interfaces.msg.FileAttributes(photo_ratio=0, photo_rotation=0, video_duration=0, video_frame_rate=0, video_rotation=0, video_resolution=0), number_sub_files=0, sub_files=[])`

After that, I called the `/wrapper/psdk_ros2/camera_download_file_by_index` action and successfully downloaded the file.

`ros2 action send_goal /wrapper/psdk_ros2/camera_download_file_by_index psdk_interfaces/action/CameraDownloadFileByIndex "{payload_index: 1, file_index: 262161, file_name: 'DJI_20240517103721_0001_T.JPG', file_path: '/home/ubuntu/'}"
`
Now that I've been able to download the picture, we can close this issue.

biancabnd, 2024-08-13:
Great to hear that! Thanks for the feedback

closed 2024-08-13 by biancabnd