umdlife/psdk_ros2 #135: Could not subscribe successfully to some topics

Janphr, 2024-09-04:
Hey,

I get these errors when using default data_frequencies:
```
[psdk_wrapper_node-1] [ERROR] [1725436823.487022008] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_STATUS_LANDINGGEAR, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.528727289] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_STATUS_MOTOR_START_ERROR, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.566005911] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_FLIGHT_ANOMALY, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.607857748] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_BATTERY_INFO, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.647292477] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_HEIGHT_FUSION, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.687609594] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_CONTROL_DEVICE, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.729046727] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_HOME_POINT_INFO, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.766345886] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_HOME_POINT_SET_STATUS, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.806795614] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_AVOID_DATA, error 236
[psdk_wrapper_node-1] [ERROR] [1725436823.847944748] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_ALTITUDE_OF_HOMEPOINT, error 236
```

topics like height_above_ground, flight_status, ... aren't published. using psdk samples, everything works. i saw in psdk that DJI_FC_SUBSCRIPTION_TOPIC_STATUS_FLIGHT is subscribed with 10 hz, so i changed flight_status to 10. after that only 3 subscritions fail:

```
[psdk_wrapper_node-1] [ERROR] [1725437463.506612610] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_STATUS_LANDINGGEAR, error 236
[psdk_wrapper_node-1] [ERROR] [1725437463.547239511] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_STATUS_MOTOR_START_ERROR, error 236
[psdk_wrapper_node-1] [ERROR] [1725437463.585395695] [dji.telemetry_node]: Could not subscribe successfully to topic DJI_FC_SUBSCRIPTION_TOPIC_FLIGHT_ANOMALY, error 236
```

strangely, height_above_ground is being published now, but not flight_status. 
i'm using m350 and most recent psdk_ros2

any idea what's going on?

best regards!

Janphr, 2024-09-04:
i commented the 3 that were still failing and now flight_status is publishing as well :)

i tried other frequencies for the 3, but no luck. 
maybe simply not available for m350?

biancabnd, 2024-09-17:
Hi @Janphr, 
You are correct. It is usually either the frequency which is too high for a certain topic (try 1 Hz if you haven't done it already), or that the topic info simply is not available for a certain device. The error code simply maps to DJI_ERROR_SYSTEM_MODULE_CODE_SYSTEM_ERROR which does not give much of a hint of what is going on. You can always disable certain topic publications by simply assigning a 0 Hz freq in the config file. 
Sorry that I cannot provide you with more information on this.

Janphr, 2024-09-18:
Hey, thanks for replying!

> You can always disable certain topic publications by simply assigning a 0 Hz freq in the config file.

Unfortunately, the 3 I commented are part of `flight_status_frequency`. 
Maybe they could be moved to their own category.

Another unrelated question:
Do you think there is any way to enable/disable `discreet mode` via psdk?

biancabnd, 2024-09-18:
Yes, there is a certain grouping of topics which could be improved indeed. 

Regarding the discreet mode, as far as I know there is no way to control this setting via the PSDK. You can check it in  DJI's repo https://github.com/dji-sdk/Payload-SDK to see if there is any mention of that but I am afraid is not something that is exposed via the psdk.

Janphr, 2024-09-18:
Yeah that's what I thought 😕
Thank you!

closed 2024-09-18 by Janphr