umdlife/psdk_ros2 #139: I'm having trouble compiling

USTBhyh, 2024-09-25:
Hello, thank you very much for the great library. But I'm having some problems compiling, can you help me with some suggestions?
When I run `colcon build`, its second ROS package fails to compile and the output looks like this：
```
CMake Error: The source directory "/home/malab/psdk_ros2_ws/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
gmake[2]: *** [samples/sample_c++/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux_cxx.dir/build.make:1073: bin/dji_sdk_demo_linux_cxx] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:170: samples/sample_c++/platform/linux/manifold2/CMakeFiles/dji_sdk_demo_linux_cxx.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2

---
Failed   <<< entry [8.70s, exited with code 2]
Summary: 2 packages finished [9.41s]
  1 package failed: entry
  1 package had stderr output: entry
```

USTBhyh, 2024-09-25:
The structure of my file is shown in the image：
[image]

USTBhyh, 2024-09-25:
oh, I know what the problem is, this error is a compilation error of my PSDK. So where should I put the PSDK files?
[image]

biancabnd, 2024-10-23:
Hi @USTBhyh, 
Just try to compile only the psdk packages using this command: `colcon build --packages-select psdk_wrapper psdk_interfaces`

USTBhyh, 2024-11-21:
Oh, well. Thanks.

closed 2024-11-21 by USTBhyh