umdlife/psdk_ros2 #147: Unreliable start of psdk_wrapper

tompe17, 2024-11-11:
I am having issues with starting psdk_wrapper sometimes failing.  I start with:
```
ros2 launch psdk_wrapper wrapper.launch.py link_config_file_path:=/home/lrs/lrs2_ws/src/lrs_m300/cfg/link_config.json psdk_params_file_path:=/home/lrs/lrs2_ws/src/lrs_m300/cfg/psdk_params.yaml namespace:=/dji5
```
and sometimes get
```
[INFO] [launch]: All log files can be found below /home/lrs/.ros/log/2024-11-11-16-34-04-460977-mat5-35889
[INFO] [launch]: Default logging verbosity is set to INFO                         
[INFO] [psdk_wrapper_node-1]: process started with pid [36441]                    
[psdk_wrapper_node-1] [INFO] [1731339244.964723431] [dji5.psdk_wrapper_node]: Creating Constructor PSDKWrapper
[psdk_wrapper_node-1] [ERROR] [1731339244.965214816] [dji5.psdk_wrapper_node]: is_waypoint_flying_module_mandatory_: 1
[psdk_wrapper_node-1] [INFO] [1731339244.979036029] [dji5.telemetry_node]: Creating TelemetryModule
[psdk_wrapper_node-1] [INFO] [1731339244.990261983] [dji5.flight_control_node]: Creating FlightControlModule
[psdk_wrapper_node-1] [INFO] [1731339245.001426849] [dji5.waypoint_flying_node]: Creating WaypointFlyingModule
[psdk_wrapper_node-1] [INFO] [1731339245.020803755] [dji5.camera_node]: Creating CameraModule
[psdk_wrapper_node-1] [INFO] [1731339245.035299744] [dji5.gimbal_node]: Creating GimbalModule
[psdk_wrapper_node-1] [INFO] [1731339245.050217265] [dji5.liveview_node]: Creating LiveviewModule
[psdk_wrapper_node-1] [INFO] [1731339245.068056590] [dji5.hms_node]: Creating HmsModule
[psdk_wrapper_node-1] [WARN] [1731339245.111480562] [rcl_lifecycle]: No transition matching 3 found for current state unconfigured
[psdk_wrapper_node-1] [ERROR] [1731339245.111513862] [dji5.psdk_wrapper_node]: Unable to start transition 3 from current state unconfigured: Transition is not registered., at ./src/rcl_lifecycle.c:355     
[psdk_wrapper_node-1] [INFO] [1731339245.112513478] [dji5.psdk_wrapper_node]: Configuring PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1731339245.112539869] [dji5.psdk_wrapper_node]: Loading parameters
[psdk_wrapper_node-1] [INFO] [1731339245.112582028] [dji5.psdk_wrapper_node]: App name: Dummy
[psdk_wrapper_node-1] [INFO] [1731339245.112591342] [dji5.psdk_wrapper_node]: App id: 153720
[psdk_wrapper_node-1] [INFO] [1731339245.112600428] [dji5.psdk_wrapper_node]: App key: ec72214329b2395951193e16af289e0
[psdk_wrapper_node-1] [INFO] [1731339245.112612476] [dji5.psdk_wrapper_node]: Baudrate: 921600
[psdk_wrapper_node-1] [INFO] [1731339245.112621928] [dji5.psdk_wrapper_node]: Using connection configuration file: /home/lrs/lrs2_ws/src/lrs_m300/cfg/link_config.json                                       
[psdk_wrapper_node-1] [INFO] [1731339245.112682278] [dji5.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1731339245.112712675] [dji5.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1731339245.112720613] [dji5.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] [INFO] [1731339245.112836682] [dji5.psdk_wrapper_node]: Loaded configuration file
[psdk_wrapper_node-1] [INFO] [1731339245.112845985] [dji5.psdk_wrapper_node]: Using DJI_USE_UART_USB_BULK_DEVICE
[psdk_wrapper_node-1] [INFO] [1731339245.112855305] [dji5.psdk_wrapper_node]: Environment has been set!
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/dji5/psdk_wrapper_node'
```

Is this a bug or have I messes up (I am working on a branch adding support for Waypoint V2 flying)?

biancabnd, 2024-11-20:
Hi @tompe17! This most probably means the time it takes for the node to transition from one state to another, in this case from configure to activate takes longer than the default timeout. And this seems to be related, with the error you get above : _No transition matching 3 found for current state unconfigured_. So, if you have added a new module, with a new node, there might be something wrong with how you transition the node through all the lifecycle states. Also, check with the main branch to see if you get the same behavior. I would expect that to work.

michele-colombo, 2025-01-21:
Hi,
I cannot find any reference to a default timeout for handling LifecycleNode state transitions. Is this something implemented in you code? Or is it something of ROS2?
Thanks!

AndreaSanchini, 2025-05-13:
Hi @tompe1,
maybe  you've already solved the issue, but I found a potential workaround. I encountered the same problem (it would fail to start once every three/four times). From my limited understanding of lifecycle nodes, the issue seems to stem from the `wrapper.launch.py` file, where we emit both `CONFIGURE` and `ACTIVATE` transition events. Since `on_configure()` can take some time to complete, the activate event might be triggered before configuration finishes, leading to the error:

```[psdk_wrapper_node-1] [WARN] [1747128038.350301876] [rcl_lifecycle]: No transition matching 3 found for current state unconfigured
[psdk_wrapper_node-1] [ERROR] [1747128038.350643415] [wrapper.psdk_wrapper_node]: Unable to start transition 3 from current state unconfigured: Transition is not registered., at ./src/rcl_lifecycle.c:355
[psdk_wrapper_node-1] [INFO] [1747128038.351712288] [wrapper.psdk_wrapper_node]: Configuring PSDKWrapper
[ERROR] [launch_ros.actions.lifecycle_node]: Failed to make transition 'TRANSITION_ACTIVATE' for LifecycleNode '/wrapper/psdk_wrapper_node'
```

I managed to fix the problem by ensuring that `ACTIVATE` event is only fired after the node has returned from `on_configure()` by wrapping it inside a `RegisterEventHandler' modifying the last part of the launch file from:

```
# Activate lifecycle node
    wrapper_activate_trans_event = EmitEvent(
        event=ChangeState(
            lifecycle_node_matcher=launch.events.matches_action(wrapper_node),
            transition_id=lifecycle_msgs.msg.Transition.TRANSITION_ACTIVATE,
        )
    )

    # Create LaunchDescription and populate
    ld = LaunchDescription()

    # Declare Launch options
    ld.add_action(declare_namespace_cmd)
    ld.add_action(declare_psdk_params_cmd)
    ld.add_action(declare_link_config_cmd)
    ld.add_action(declare_hms_codes_cmd)
    ld.add_action(wrapper_node)
    ld.add_action(wrapper_configure_trans_event)
    ld.add_action(wrapper_activate_trans_event)
```

to: 

```
# Activate lifecycle node
    wrapper_activate_trans_event = EmitEvent(
        event=ChangeState(
            lifecycle_node_matcher=launch.events.matches_action(wrapper_node),
            transition_id=lifecycle_msgs.msg.Transition.TRANSITION_ACTIVATE,
        )
    )

    # When wrapper node reaches 'inactive' state (configuration finished), make it activate
    wrapper_inactive_state_handler = RegisterEventHandler(
        OnStateTransition(
            target_lifecycle_node=wrapper_node,
            goal_state='inactive',
            entities=[
                wrapper_activate_trans_event
            ],
        )
    )

    # Create LaunchDescription and populate
    ld = LaunchDescription()

    # Declare Launch options
    ld.add_action(declare_namespace_cmd)
    ld.add_action(declare_psdk_params_cmd)
    ld.add_action(declare_link_config_cmd)
    ld.add_action(declare_hms_codes_cmd)
    ld.add_action(wrapper_inactive_state_handler)
    ld.add_action(wrapper_node)
    ld.add_action(wrapper_configure_trans_event)
```

Please note that I have very little experience with managing lifecycle nodes, but in my case, this fix made the startup of the wrapper completely reliable.