umdlife/psdk_ros2 #152: perception module stops working when rotating through directions

Janphr, 2024-12-04:
Hey,

Thanks for the perception module @bitcurious!
I want to use all directions, so I wrote a simple routine that starts one direction, waits for both images and params, and then switches to the next, which seemed to work fine at first.
Unfortunately, the module stops working after a few minutes and it stops publishing images.

I can still send service calls and it still returns success, except for the last direction before it stopped working:
`Could not start the perception stereo vision stream. Error code: 236` 
I tried stopping or starting a different direction, but it does not recover. 
Only restarting psdk helps. 
Any idea what could be causing this? I'm not spamming service calls or anything...
Running only one direction works fine.

M350 with Jetson Nano Orin on jp6.1 (Ubuntu 22.04.5), ROS2 Humble

Janphr, 2024-12-04:
Update:
I simply subscribed to all once, which seems to run stable. Maybe something you could add. E.g. as mode ALL in direction_map_

closed 2024-12-04 by Janphr

bitcurious, 2024-12-09:
@Janphr Thanks for update, I will try it and let you know. Also for current perception module which I added I wanted to have only one direction at time with both side, for me it was working fine.