umdlife/psdk_ros2 #154: Mavic3T Compatibility

JaMateoslu, 2025-01-27:
Hi all, first of all thanks for this amazing repo. 

Following the topic of this Issue, I'd like to know whether the Mavic3/3T or any drones from the series have been tested with this repo with the DJI expansion board. I am planning on purchasing the Expansion Board and wanted to know beforehand if this worked on the Mavic Series.

https://store.dji.com/es/product/dji-e-port-development-kit?vid=141171

I will be testing it in the upcoming weeks and posting issues/problems related to it, but first I wanted to see whether someone has tested it before on the Mavic series and could provide advice about it (in case you have used Jetson Nnao or Raspberry too).

Thanks in advance and I will keep you posted,

biancabnd, 2025-01-28:
Hi @JaMateoslu!
I'm glad you find the repo useful. Regarding your question, we have not tested it with the Mavic3/3T series. Perhaps someone else from the community has but I am not aware of it. I think it's great that you want to try it out and please do post here your findings. 🚀

bitcurious, 2025-05-30:
@JaMateoslu is this wrapper working with Mavic 3 series ? Have you tested it ?