umdlife/psdk_ros2 #158: DJI core could not be initiated. Error code is: 236

AreteQin, 2025-02-22:
Hi there

Thank you for sharing your excellent work!

I am using the following configuration:
Drone: M30T
Onboard computer: nvidia jetson xavier nx
Payload-SDK: https://github.com/whyscience/Payload-SDK.git -b V3.8.1
PSDK-ROS2: https://github.com/umdlife/psdk_ros2 -b main

link_config.json:
```
        "link_select": "use_uart_and_usb_bulk_device",
        "uart_config": {
            "uart1_device_name": "/dev/ttyTHS1",
            "uart2_device_enable": "true",
            "uart2_device_name": "/dev/ttyTHS1"
        },
        "network_config": {
            "network_device_name": "enxf8e43b7bbc2c",
            "network_usb_adapter_vid": "0x0B95",
            "network_usb_adapter_pid": "0x1790"
        },
```
psdk_params.yaml:
```
#define USER_APP_NAME               "hidden"
#define USER_APP_ID                 "hiden"
#define USER_APP_KEY                "hidden"
#define USER_APP_LICENSE            "hidden"
#define USER_DEVELOPER_ACCOUNT      "hidden"
#define USER_BAUD_RATE              "1000000"
```
I have tested the sample "./dji_sdk_demo_linux_cxx" in Payload-SDK and got the following output:
```
[0.027][core]-[Info]-[DjiCore_Init:105) Payload SDK Version : V3.8.1-beta.0-build.2006 
[3.313][adapter]-[Info]-[DjiAccessAdapter_Init:231) Identify aircraft series is Matrice 30 Series 
[3.313][adapter]-[Info]-[DjiAccessAdapter_Init:264) Identify mount position type is Extension Port Type 
[3.352][adapter]-[Info]-[DjiAccessAdapter_Init:365) Identity uart0 baudrate is 1000000 bps 
[5.362][adapter]-[Info]-[DjiPayloadNegotiate_Init:219) Waiting payload negotiate finish. 
[6.593][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:474) Updating dji sdk policy file... 
[7.593][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:482) Update dji sdk policy file successfully 
[7.606][core]-[Info]-[DjiCore_Init:173) Identify AircraftType = Matrice 30T, MountPosition = Extension Port, SdkAdapterType = None 
[8.130][user]-[Info]-[SetVolume:580) Set widget speaker volume: 60 
[8.130][user]-[Warn]-[SetVolume:598) No audio device found, please add audio device and init speaker volume here!!! 
[14.633][core]-[Info]-[DjiCore_ApplicationStart:230) Start dji sdk application 
[14.633][user]-[Info]-[DjiUser_ApplicationStart:363) Application start. 

| Available commands:                                                                              |
| [0] Fc subscribe sample - subscribe quaternion and gps data                                      |
| [1] Flight controller sample - you can control flying by PSDK                                    |
| [2] Hms info manager sample - get health manger system info by language                          |
| [a] Gimbal manager sample - you can control gimbal by PSDK                                       |
| [c] Camera stream view sample - display the camera video stream                                  |
| [d] Stereo vision view sample - display the stereo image                                         |
| [e] Run camera manager sample - you can test camera's functions interactively                    |
| [f] Start rtk positioning sample - you can receive rtk rtcm data when rtk signal is ok           |
```
However, when I try to run the psdk-ros2 using `ros2 launch psdk_wrapper wrapper.launch.py`. I got the following error:
```
...
[psdk_wrapper_node-1] [INFO] [1740256055.493691674] [wrapper.psdk_wrapper_node]: Baudrate: 1000000
[psdk_wrapper_node-1] [INFO] [1740256055.493769850] [wrapper.psdk_wrapper_node]: Using connection configuration file: /home/qin/foxy_ws/install/psdk_wrapper/share/psdk_wrapper/cfg/link_config.json
[psdk_wrapper_node-1] [INFO] [1740256055.494046458] [wrapper.psdk_wrapper_node]: Setting environment
[psdk_wrapper_node-1] [INFO] [1740256055.494242331] [wrapper.psdk_wrapper_node]: Registered OSAL handler
[psdk_wrapper_node-1] [INFO] [1740256055.494331963] [wrapper.psdk_wrapper_node]: Registered HAL handler
[psdk_wrapper_node-1] [INFO] [1740256055.495161148] [wrapper.psdk_wrapper_node]: Loaded configuration file
[psdk_wrapper_node-1] [INFO] [1740256055.495234268] [wrapper.psdk_wrapper_node]: Using DJI_USE_UART_AND_NETWORK_DEVICE
[psdk_wrapper_node-1] [INFO] [1740256055.495298908] [wrapper.psdk_wrapper_node]: Environment has been set!
[psdk_wrapper_node-1] [INFO] [1740256055.500653796] [wrapper.psdk_wrapper_node]: Activating PSDKWrapper
[psdk_wrapper_node-1] [INFO] [1740256055.500805924] [wrapper.psdk_wrapper_node]: Init DJI Core...
[psdk_wrapper_node-1] enxf8e43b7bbc2c: ERROR while getting interface flags: No such device
[psdk_wrapper_node-1] [ERROR] [1740256060.237920655] [wrapper.psdk_wrapper_node]: DJI core could not be initiated. Error code is: 236. Retrying for 0 time. 
```

I would greatly appreciate any help or information in advance!

mjohenneken, 2025-03-11:
Hi there, we are experiencing this issue as well on M300 RTK with Jetson Orin NX. Our workaround currently is to shutdown both drone and jetson and turn the on again. In our case the issue happens the second time we try to run the psdk_wrapper.

AreteQin, 2025-03-12:
Hi, thank you for your hints!

closed 2025-03-12 by AreteQin